slic3r/Slic3r #2698: default extrusion width set to "auto" not working, works if set at 100%

brunofoc, 2015-03-01:
I sliced a small model with slic3r 1.2.6;
using 0.2mm layer height 
setting advanced->default extrusion width to 0 (auto) generates strange settings:
`
 generated by Slic3r 1.2.6 on 2015-03-01 at 13:27:39

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.85mm
; infill extrusion width = 1.02mm
; solid infill extrusion width = 0.85mm
; top infill extrusion width = 0.85mm
`
[image]

setting advanced->default extrusion width to 100% generates the correct settings:

`
; generated by Slic3r 1.2.6 on 2015-03-01 at 13:28:45

; external perimeters extrusion width = 0.20mm
; perimeters extrusion width = 0.20mm
; infill extrusion width = 0.20mm
; solid infill extrusion width = 0.20mm
; top infill extrusion width = 0.20mm
`
[image]

the model (manifold) is attached here

alexrj, 2015-03-01:
What do you mean by "strange settings" and "not working"?
Can you please tell us why do you think those values are "strange" and "not working"?

alexrj, 2015-03-01:
Also, have you read the guidelines before posting this issue?

brunofoc, 2015-03-01:
which part of the guidelines I am missing? sorry it's the first time I post here.

this is a preview of the model:
[image]

The first  picture of the g-code preview of layer 33 shows that the perimeter is not closed, and lacks of the honeycomb infill;
( I also noticed that  infill extrusion width = 1.02mm , wich is big, but maybe is a correct behaviour)

The second g-code preview shows the correct version.

the only difference is using  advanced->default extrusion width to 0 in the first and to 100% in the second

my .inf file is this:
https://drive.google.com/file/d/0B_WZ_InWYbxacklxS1U5MzNzTjg/view?usp=sharing
I used the settings  RepRap 0.2mm, RepRap Pla Blu 3mm, Reprap 20x20cm

maybe i am missing something about the slic3 settings, or there is a bug in the Repetier Host g-code preview.

thank you

alexrj, 2015-03-01:
You haven't explained why the automatic extrusion widths are "strange".
Actually, they're not strange at all. Trying to print with a 0.2mm extrusion width using a 0.5mm nozzle, as you're trying to do, is very wrong.

You should use Slic3r's embedded G-code preview. We can't help with other previewers.

closed 2015-03-01 by alexrj

brunofoc, 2015-03-01:
Ok, I catched it , using not coherent values may lead to unpredictable results;
i use to print at 0.2mm with good results, maybe the problem arises with very small details;

therefore i tried to slice  the model with reprap 0.3mm (coherent with the nozzle diameter), and there still appear some artefacts:
[image]

when i change the default extrusion width to 100%, the gcode preview seems ok:
[image]

maybe the issue is related to print object with very small details on a big nozzle; slic3r  could warn about uncorrect settings, or switch to fixed extrusion width

brunofoc, 2015-03-01:
I also made a small resarch about nozzle diameter/ resolution, and in this article RichRap states that he use 0.5 mm nozzle at 0.2mm resolution (at the end of the page, "Summary and general tips and tricks ")

http://richrap.blogspot.it/2015/01/slic3r-advanced-perimeter-tuning-3d.html

a4jp-com, 2015-03-02:
If we have a 0.4 or 0.5 nozzle what size extrusion do you recommend?

brunofoc, 2015-03-02:
Ok I found my mistake: 
the layer height and the  extrusion width are two different parameters, so with a 0.5mm nozzle it is possibile to print a 0.2mm layer height, but it is wrong to use a 0.2 mm extrusion size;

the recomended default extrusion size , as pointed  out [in this RichRap post ("Summary and general tips and tricks") ](http://richrap.blogspot.it/2015/01/slic3r-advanced-perimeter-tuning-3d.html), is about 100%-150% the nozzle size; so  for a 0.5mm nozzle size the default extrusion width should be about 0.5-0.75mm; better solution is leave it Auto in the advanced settings; so the generated settings above are not strange.

Using auto settings  I still have holes in the perimeter as pointed before;
this is probably due to the thin perimeter walls, smaller than  the perimeter size (~0.50mm)
[image]

thank you

alexrj, 2015-03-02:
Well yes, with a 0.5mm it's quite common to have small gaps in detailed models as not all narrow spaces can be filled properly...

lordofhyphens, 2015-03-02:
Yeah, kinda need variable width paths but that covering problem is
intractable. :(

gordo3di, 2015-03-13:
We have been racking our brains over this for a few weeks now. I can confirm something has changed in 1.1.7. We have recently been reworking our config files and for some reason when setting the default extrusion widths to 0 weird things do indeed happen.

We have been noticing our infill and our top solid infill has been printing VERY thin. We have literally spent 2 weeks testing EVERYTHING on our printers to diagnose the problem.

Of course today I finally decided to open the gcode and the problem was right in front of me.

Here is the gcode header from an older file:
; generated by Slic3r 1.1.7 on 2014-12-18 at 11:28:56

; perimeters extrusion width = 0.50mm
; infill extrusion width = 0.79mm
; solid infill extrusion width = 0.79mm
; top infill extrusion width = 0.79mm

Here is the gcode header from a newly sliced file:
; generated by Slic3r 1.1.7 on 2015-02-24 at 14:15:19

; perimeters extrusion width = 0.50mm
; infill extrusion width = 0.30mm
; solid infill extrusion width = 0.84mm
; top infill extrusion width = 0.23mm

As you can see for some reason the infill and solid infill values are much lower in the second file causing the part to be very brittle and look bad. It looks like this happened because I lowered the top infill to 90% instead of 0 which means it will calculate based on the layer height. I did not do this for the general infill but setting the top infill back to 0 seemed to fix the problem.

Just putting it here so other people know.