slic3r/Slic3r #2699: Environment variables in post-processing scripts

hippich, 2015-03-01:
I wrote small post-processing script - https://github.com/hippich/gcode-postprocessing/blob/master/gradual-bed-temp.pl

originally I wanted to use SLIC3R_ environment variables to use first and second layer bed temperature as options for start_temp and end_temp, but I found that %ENV variable is always contain only system environment vars and none of SLIC3R_*

Is there something I need to check to expose these?

alexrj, 2015-03-02:
I just tested with the following test script and %ENV is populated:

```perl
#!/usr/bin/perl

use XXX;
YYY \%ENV;
```

What's your operating system? Are you using the binary Slic3r or running from git source?

hippich, 2015-03-02:
I just used exact same example and got only system env variables.. thing is - I am running it under sudo, since without sudo it breaks - perhaps something about my config. I am wondering, may be there is something about sudo preventing env variables?

I am on Ubuntu 14.04 64bit, master checkout from github.com.

Error I see if I run it without sudo:

```
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 1141 error_code 8 request_code 66 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

alexrj, 2015-03-02:
I have no idea, but you can try to slice from command line so that GUI is not launched and you don't get that error. That way you can see whether not using sudo would prevent the environment variables issue...

hippich, 2015-03-02:
uh, so obvious... i don't know why i did not try that. `sudo` suspicion confirmed - it does not pass all ENV variables when run with `sudo` and works as expected otherwise. Thank you and sorry for time wasted.

ps: should I submit pull request with this post processing script or post processing scripts are outside of scope of this repository?

hippich, 2015-03-04:
@alexrj, so i think i was too quick with closing this issue. With my workaround from #2704 I was able to run GUI and try same post-processing script. None SLIC3R_* variables available when run via GUI. And all variables are there when run from command-line with the same post-processing script. Any clue?

hippich, 2015-03-05:
Digged more today and I can confirm %ENV is set properly before running system() call. So this is probably something about my shell, although still not sure where to look to confirm it.

lordofhyphens, 2016-07-12:
If you're starting the gui from command line, anything set in your shell should be available. Otherwise the variable would likely have to be set in your logon profile to be visible to Slic3r.

closed 2016-07-12 by lordofhyphens