slic3r/Slic3r #2700: printer "has heated bed" option.

whosawhatsis, 2015-03-01:
It would be nice to have an option in the printer profile to specify whether the machine has a heated bed. With this option unchecked, all M140/M190 codes would be left out of the gcode, but the same  filament profiles could still be used for a different machine that does have a heated bed to produce those codes.

alexrj, 2015-03-02:
Will do.  :)

jonaskuehling, 2015-03-04:
Apologize my dumb question :) but does it hurt at all, if M140/M190 commands are sent to a non-heatbed-printer? Does it try to heat up infinitely then?

Never tried actually :)

alexrj, 2015-03-04:
Unfortunately yes!  (Or so it was last time it happened to me)

whosawhatsis, 2015-03-05:
M140 doesn't hurt, but M190 makes it wait for something that will never happen before proceeding, effectively locking up the printer.

kefir-, 2015-03-05:
Shouldn't the firmware handle (ignore) M190 if there's no heatbed?

jonaskuehling, 2015-03-05:
That's at least what I would have expected...

lordofhyphens, 2016-07-13:
There you go.

alexrj, 2016-12-16:
Merged, we have a "Has heatbed option".

closed 2016-12-16 by alexrj