slic3r/Slic3r #2701: Request 1.2.7-dev: When raft layers >0, allow first layer height to be equal to raft-extruder-nozzle-diameter

jonaskuehling, 2015-03-02:
Hi Alex,

thanks a bunch for fixing #2662! Now that we have almost full control over the very first raft/support layer I'd like to suggest one further minor detail:

In case "raft layers" > 0 it would be awesome to be able to set the first layer height equal to the support material extruder nozzle diameter at maximum. So, when printing with Ø0.35mm nozzle on the first extruder (model material) and e.g. Ø0.5mm nozzle on the second extruder (support material), one could use up to 0.5mm height for the first layer and thus achieve further redundancy in regard to bed leveling/evenness.

Currently first layer height only seems to be allowed up to the size of the smallest nozzle installed on one of the used extruders - which is of course mandatory when using both (or more) extruders in the first layer already. But I think for raft>0 Slic3r should only check the nozzle size of the extruder, that is selected for "Support material/raft/skirt", since in this case there will always be only this single extruder printing the first layer directly on the print bed.

What do you think?

alexrj, 2015-03-02:
That makes sense, I'll check how much work is needed for that. :)

alexrj, 2015-03-06:
Done! :)

closed 2015-03-06 by alexrj

jonaskuehling, 2015-03-07:
Works like a charm! ;-)