slic3r/Slic3r #2703: Wipe + complete individual objects causes head crash

sorki, 2015-03-02:
Seems like last retract_wipe is repeated once again after move to origin for next object which causes head crashing into finished object.

Slic3r 1.2.7-dev (latest head).

```
G1 Z8.000 F9000.000 ; lift Z
G1 F7200
G1 X17.998 Y-13.305 E-0.15446 ; retract;_WIPE
G1 F7200
G1 X17.047 Y-13.305 E-0.49314 ; retract;_WIPE
G1 F7200
G1 X18.305 Y-12.047 E-1.12656 ; retract;_WIPE
G1 F7200
G1 X17.706 Y-11.695 E-1.37397 ; retract;_WIPE
G1 F7200
G1 X16.097 Y-13.305 E-2.18476 ; retract;_WIPE
G1 F7200
G1 X15.146 Y-13.305 E-2.52344 ; retract;_WIPE
G1 F7200
G1 X15.794 Y-12.656 E-2.85000 ; retract;_WIPE
G92 E0 ; reset extrusion distance
G1 X-20.000 Y-15.000 F9000.000 ; move to origin position for next object
M104 S205 ; set temperature
M107 ; disable fan
G1 F7200
G1 X17.998 Y-13.305 E-0.15446 ; retract
G1 F7200
G1 X17.047 Y-13.305 E-0.49314 ; retract
G1 F7200
G1 X18.305 Y-12.047 E-1.12656 ; retract
G1 F7200
G1 X17.706 Y-11.695 E-1.37397 ; retract
G1 F7200
G1 X16.097 Y-13.305 E-2.18476 ; retract
G1 F7200
G1 X15.146 Y-13.305 E-2.52344 ; retract
G1 F7200
G1 X15.794 Y-12.656 E-2.85000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.200 F9000.000 ; move to next layer (0)
G1 Z1.200 F9000.000 ; lift Z
```

Fine with wipe disabled:

```
G1 X18.305 Y-12.998 E4.27980 ; infill
G1 E1.27980 F2700.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z8.000 F9000.000 ; lift Z
G1 X-20.000 Y-15.000 F9000.000 ; move to origin position for next object
M104 S205 ; set temperature
M107 ; disable fan
G1 Z0.200 F9000.000 ; move to next layer (0)
G1 Z1.200 F9000.000 ; lift Z
G1 X-19.086 Y-14.086 F9000.000 ; move to first perimeter point
```

c-born, 2015-03-02:
Looks very similar to #2691

alexrj, 2015-03-02:
Aha, so wipe is the culprit. Will fix it.

alexrj, 2015-03-06:
I pushed a fix. @sorki, can you confirm the issue is gone?

sorki, 2015-03-08:
Yup. It's gone. Thank you!

closed 2015-03-08 by sorki