slic3r/Slic3r #2704: X Window System error - BadMatch (invalid parameter attributes)

hippich, 2015-03-02:
Version of Slic3r with 3D viewer now gives me following X Window error when run as normal user. This does not happen when run with `sudo`. I believe there is something wrong with X configuration, but I am not sure where to start debugging it. Running under Ubuntu 14.04 64bit. Any ideas are very welcome:

```
 ./slic3r.pl                                                                                                                                                                                                                
(slic3r.pl:24925): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16

(slic3r.pl:24925): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16

(slic3r.pl:24925): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 1141 error_code 8 request_code 66 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

```
  uname -a                                                                                                                                                                                                                   Linux joe-pc 3.17.8-031708-generic #201501081837 SMP Thu Jan 8 18:39:03 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

hippich, 2015-03-02:
After googling and trying random things, I found that this will run it:

$ XLIB_SKIP_ARGB_VISUALS=1 ./slic3r.pl

question is - is this is bug in Slic3r or it is problem in my system?

alexrj, 2015-03-02:
For sure it's not in Slic3r, but I can't say whether it's a problem in your system or some issue in the wxGTK / GTK / X11 toolchain upon which Slic3r relies on Linux…

closed 2015-03-02 by hippich