slic3r/Slic3r #2705: 1.2.6 crash on mac

alan412, 2015-03-03:
This is very bizarre.   1.2.6 was working great for me.   Now, every time it gets to the end of slicing (whether in background or manual) it gets to the end of the scroll bar and crashes.  From console.log I don't see anything except "Service exited with abnormal code: 255"

:-(

alan412, 2015-03-03:
It had to do with settings in slic3r.ini.  It is very bizarre, by getting rid of the file I could get back working.   But putting back the "bad.ini" doesn't cause it to crash again.....   hmmmm.

alan412, 2015-04-01:
Ok, now this is strange.  It crashes every time for me, but I can get it to work by the following steps:
1) Delete ~/Library Application Support/Slic3r/slic3r.ini
2) Start up and choose "expert mode"
3) Exit out and restart
4) Elect filament, print, and printer.   It will make the STL files fine until I exit out of Slic3r.  Next time out, it crashes again when it gets to the end of slicing until I do steps 1-3 again.

alan412, 2015-04-01:
I even removed the application (backing up the filament, print, and printer directories and restoring them) with the same problem.  Very bizarre.  I used AppZapper which had me delete the plist file, the ini files, the application, etc.

alan412, 2015-04-04:
Perhaps this will help.   I started it with --debug and I get the following:
Attempt to free unreferenced scalar: SV 0xd97b010, Perl interpreter: 0x24b5c00 at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 116.

Here is all of the output:
MacOS/$ ./slic3r --debug
wxWidgets version wxWidgets 3.0.0, Wx version 0.9923
Data directory: /Users/alan/Library/Application Support/Slic3r
Checking for updates...
Attempt to free unreferenced scalar: SV 0x2739200, Perl interpreter: 0x175ee00 at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r/GUI.pm line 262.
Attempt to free unreferenced scalar: SV 0x27391c0, Perl interpreter: 0x175ee00 at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r/GUI.pm line 262.
2015-04-04 16:42:27.847 slic3r[23995:1339641] Unable to simultaneously satisfy constraints:
(
    "<NSAutoresizingMaskLayoutConstraint:0xc25c9d0 h=--& v=--& V:[FI_TBrowserBackgroundView:0xc271700(0)]>",
    "<NSAutoresizingMaskLayoutConstraint:0xc25a0f0 h=-&- v=-&- V:|-(0)-[FI_TListScrollView:0xd353710]   (Names: '|':FI_TBrowserBackgroundView:0xc271700 )>",
    "<NSAutoresizingMaskLayoutConstraint:0xc25a0b0 h=-&- v=-&- V:[FI_TListScrollView:0xd353710]-(0)-|   (Names: '|':FI_TBrowserBackgroundView:0xc271700 )>",
    "<NSAutoresizingMaskLayoutConstraint:0xc2715d0 h=-&- v=-&- V:[NSClipView:0xd353510]-(15)-|   (Names: '|':FI_TListScrollView:0xd353710 )>",
    "<NSAutoresizingMaskLayoutConstraint:0xc26ad30 h=-&- v=-&- V:|-(0)-[NSClipView:0xd353510]   (Names: '|':FI_TListScrollView:0xd353710 )>"
)

Will attempt to recover by breaking constraint 
<NSAutoresizingMaskLayoutConstraint:0xc2715d0 h=-&- v=-&- V:[NSClipView:0xd353510]-(15)-|   (Names: '|':FI_TListScrollView:0xd353710 )>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
Background processing started.
Starting thread 3 (parent: 0)...
Starting thread 4 (parent: 3)...
Making perimeters for layer 0
Making perimeters for layer 1
Making perimeters for layer 2
Making perimeters for layer 3
Making perimeters for layer 4
Making perimeters for layer 5
Making perimeters for layer 6
Making perimeters for layer 7
Making perimeters for layer 8
Making perimeters for layer 9
Making perimeters for layer 10
Making perimeters for layer 11
Making perimeters for layer 12
Making perimeters for layer 13
Making perimeters for layer 14
Starting thread 5 (parent: 3)...
Starting thread 6 (parent: 3)...
Starting thread 7 (parent: 3)...
Detecting solid surfaces...
  layer 0 has 1 bottom, 0 top and 0 internal surfaces
  layer 1 has 0 bottom, 0 top and 1 internal surfaces
  layer 2 has 0 bottom, 0 top and 1 internal surfaces
  layer 3 has 0 bottom, 0 top and 1 internal surfaces
  layer 4 has 0 bottom, 0 top and 1 internal surfaces
  layer 5 has 0 bottom, 0 top and 1 internal surfaces
  layer 6 has 0 bottom, 0 top and 1 internal surfaces
  layer 7 has 0 bottom, 0 top and 1 internal surfaces
  layer 8 has 0 bottom, 0 top and 1 internal surfaces
  layer 9 has 0 bottom, 1 top and 5 internal surfaces
  layer 10 has 0 bottom, 0 top and 5 internal surfaces
  layer 11 has 0 bottom, 0 top and 5 internal surfaces
  layer 12 has 0 bottom, 0 top and 5 internal surfaces
  layer 13 has 0 bottom, 0 top and 5 internal surfaces
  layer 14 has 0 bottom, 5 top and 0 internal surfaces
==> DISCOVERING HORIZONTAL SHELLS
Layer 0 has bottom surfaces
  looking for neighbors on layer 1...
    1 internal-solid and 0 internal surfaces found
Layer 9 has top surfaces
  looking for neighbors on layer 8...
    1 internal-solid and 1 internal surfaces found
Layer 14 has top surfaces
  looking for neighbors on layer 13...
    5 internal-solid and 0 internal surfaces found
Starting thread 8 (parent: 3)...
Filling layer 0:
Filling layer 1:
Filling layer 2:
Filling layer 3:
Filling layer 4:
Filling layer 5:
Filling layer 6:
Filling layer 7:
Filling layer 8:
Filling layer 9:
Filling layer 10:
Starting thread 9 (parent: 3)...
Filling layer 11:
Filling layer 12:
Filling layer 13:
Filling layer 14:
Starting thread 10 (parent: 3)...
Starting thread 11 (parent: 3)...
Background processing completed.
Attempt to free unreferenced scalar: SV 0xd97b010, Perl interpreter: 0x24b5c00 at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 116.
MacOS/$

MrRocketman, 2015-04-25:
I am also getting a crash anytime I try to slice anything.

Getting the same error when running with --debug

`
Background processing completed.
Attempt to free unreferenced scalar: SV 0x10104b70, Perl interpreter: 0x20aa400 at /Users/James/Downloads/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 116.
`

I believe the issue has something to do with Slicer 1.1.7 and 1.2.6 living on the same computer. I'm fairly certain I had opened up 1.1.7 after having been using 1.2.6 for a while and now all heck has broken loose :)

A [link] (http://mrrocketman.com/files/public/index.php?path=other%2FSlic3rSettings/) to my Slicer settings

alexrj, 2015-06-02:
@MrRocketman, unfortunately your website is not accessible anymore.
Can you try with 1.2.7 and tell whether you still have the problem?

@alan412, can you provide the config that causes the crash? Or can you test with 1.2.7 and tell whether we can close this one now? Thank you!

alan412, 2015-06-05:
I can't duplicate the issue on 1.2.7, so I am closing it.

closed 2015-06-05 by alan412