slic3r/Slic3r #2706: Request 1.2.7-dev: Provide option "first layer height = maximum possible" (i.e. = nozzle diameter)

jonaskuehling, 2015-03-04:
Hi,

one thought that's in my mind for a while now: It would be handy to have an additional option for the "first layer height" parameter, that automatically uses the configured nozzle diameter value. Think of it like a "use maximal possible layer height for first layer" functionality...

Currently the possible first layer height (print settings) heavily depends on the set nozzle diameter (printer settings) and when switching between different nozzle diameters, both the print and printer settings need to be modified - as long as I want take advantage of the maximum possible first layer height for raft... (see #2701 and #2662)

Perhaps this could be implemented as "0" option? (first layer height = 0 for automatic maximum handling, i.e. first layer height equal to nozzle diameter)?

whosawhatsis, 2015-03-04:
0 should work, though it might make more sense to implement it like the "combine infill every X layers" feature, where you can just put in an absurdly high value and it will lower it if necessary.

jonaskuehling, 2015-03-04:
Mmh..Good point, makes me think of this kind of implementation for general layer height as well... If you usually print 0.4mm layer resolution with Ø0.5mm nozzle and then switch nozzles to like Ø0.3mm, the print settings wouldn't be broken immediately but just sliced with maximal possible layer height, which ever value is smaller, nozzle or configured layer height..

The finally used layer height should be written to the top of the GCODE then, like the extrusion widths.

alexrj, 2015-03-06:
I agree with your idea. I'll do this in 1.3.x because it requires a fairly large refactoring and I don't want to break things now. :)

alexrj, 2015-03-06:
Basically, this will limit layer height to the **smallest** nozzle diameter in case of multiple extruders.

alexrj, 2016-12-21:
Done!

closed 2016-12-21 by alexrj