slic3r/Slic3r #2708: 1.2.6 Crash when parameters changed fast while preview updated

DanNsk, 2015-03-04:
Hi,

Trying to change several parameters affecting gcode one following another - if i'm not waiting for preview to be updated after each parameter, that crashes application (even not crashes - it just disappears).

Doesn't matter how complex the object - even for just cube 2x2 cm.

Win 7 x64

justmyopinion, 2015-03-05:
Probably same as #2676

MikeFitz73, 2015-03-06:
maybe we could add a feature that allows us to start the preview update after we are finished making edits to profile.... LIke and update now button.

justmyopinion, 2015-03-06:
@MikeFitz73 
You may already have that feature in preferences: "background processing" which can be unselected :-)

lordofhyphens, 2016-07-12:
@DanNsk Please check to see if this is still a problem in 1.2.9a (with or without background processing turned on). Closing for now.

closed 2016-07-12 by lordofhyphens