slic3r/Slic3r #2710: slic3r does not start.

lonelymyp, 2015-03-05:
slicer 1.2.6, win7 64. 
Some problems with my username?

[image]

hippich, 2015-03-05:
just a side note - but just because you can use cyrillic in path names (and account names,) doesn't mean you absolutely have to :)

alexrj, 2015-03-06:
I confirm there are issues with non-ASCII paths on Windows. It's a bit of a pain, but it's my current highest priority.

alexrj, 2015-03-09:
Okay, I think this commit fixes it.

Naesstrom, 2015-03-23:
Great, I noticed I had the same problem when trying the 1.2.6 version today. Any idea when 1.2.7 will be out for testing?

alexrj, 2015-05-24:
@Naesstrom, I released 1.2.7 today. Can you confirm it fixes the issue? Thank you!

closed 2015-06-20 by alexrj

bubnikv, 2017-02-17:
See the resolution to https://github.com/prusa3d/Slic3r/issues/133