slic3r/Slic3r #2712: Infill problems

rsatue, 2015-03-05:
Hi, 

I am doing some checks on what would happen with a large format 3D printer (2m x 1m x 1m) 
and when generating a test gcode to check the print time and how different options would affect I've found an issue with the infill generation. 

I am doing the tests with the following part. The part was originally 4 mm thick. I've tried to increase the thickness to 8 mm but the problem is the same. 

[image]

The part STL can be downloaded here: https://db.tt/PFqGvRP8

It seems that slic3r is doing well the perimeters but not the infill. This is the result with infill at 100%. 

[image]

Different versions of slic3r (1.2.1 and 1.2.6) are generating the same infill.

The results with other infill densities are very similar. The only way to get rid of this is with 0% infill. 

I am working on Windows 8.1-64

alexrj, 2015-03-06:
With large coordinates Slic3r overflows. I haven't' calculated what's the limit. A workaround is to switch millimeters with centimeters in your context (add a 10x factor to your firmware steps).

rsatue, 2015-03-06:
Thanks for the reply and for the suggestion! I've already tried and it works well when everything is in cm. 

Do you plan to change any setting in the next releases to work with large coordinates?

alexrj, 2015-03-07:
I don't know. Of course it would be nice to handle larger machines, but I don't want to put any performance penalty on normal usage cases.

closed 2015-03-11 by rsatue