slic3r/Slic3r #2713: [Feature Request] Allow decimal places in X/Y/Z-scaling factor

simonkuehling, 2015-03-06:
Hi Alessandro,

we are just carrying out a series of measurements on the effects of linear thermal expansion on dimensional accuracy of printed objects - to account for ABS model shrinkage after cooling down from a heated bed (110°C) and/or a heated chamber.

For this type of shrinkage compensation it would be great to be able to provide decimal places for the X/Y scaling factors in the plater. As we talk about less than 1% (like 0.7-0.85%) upscaling factors, these can currently only be approximated by rounding to full hundredths.

Simon

gege2b, 2015-03-06:
Hi

It's a duplicate of #2711

simonkuehling, 2015-03-06:
Hi @gege2b - i have to disagree with your comment.

If i understand your issue correctly, you propose an "absolute target dimension" value additionally to the proportional scaling factor currently implemented.

What i highlighted is, that the proportional scaling factor only accepts whole-number percentage values (like <code>112%</code>, but not <code>112.47%</code>) - which i'd love to use for precise upscaling of objects especially in X/Y direction to compensate temperature shrinkage when printing in a 70°C heated build chamber.

gege2b, 2015-03-06:
Woops, you're right, I've read your text a bit too fast, sorry about that

anyway, its not my issue :)

alexrj, 2015-03-09:
It's duplicate of #1617 :o)

closed 2015-03-09 by alexrj