slic3r/Slic3r #2714: poor bridging behaviour when manual supports are added

Ghoughpteighbteau, 2015-03-06:
I've added support structures as part of the print in this lego piece to make it print reliably, but slic3r tries to get clever with pinning and all this other stuff, and it turns into a mess!

[image]

https://www.dropbox.com/s/darjs2byyex5zis/testbricklego2.stl?dl=0
https://www.dropbox.com/s/u06sb01oz1r89k3/testbricklego.stl?dl=0

I don't see a way to get slic3r to produce workable gcode in this instance. 

Made with this [version of master](https://github.com/alexrj/Slic3r/tree/9332c2179166e93ab2f2fca0a1fa9099f845699c) (latest at time of writing)

lordofhyphens, 2017-03-19:
Still a problem, it appears.
[image]

Reproduction testcase:
[2714.zip](https://github.com/alexrj/Slic3r/files/852876/2714.zip)

lordofhyphens, 2017-03-19:
I suspect it has something to do with either thin walls or not being more aggressive about combining bridges (in cases like these, should probably be trying to form one big bridge over all of the outer contours, where we know we have anchor points).