slic3r/Slic3r #2715: retract goes with acceleration that was set the last time. not with the maximum

lonelymyp, 2015-03-06:
slic3r 1.2.6
When I set acceleration, it as well affects retract.
I'm set for the perimeter low acceleration and retract done slowly than might be.
according to the g-code acceleration for perimeter placed before going retract. 

example:
[image]

alexrj, 2015-03-13:
AFAIK, the `M204 Sx` command does *not* affect retractions (at least in Marlin). According to Marlin documentation, acceleration for retract moves is set using `M204 Tx`.

alexrj, 2015-03-13:
Also, there's an error in your screenshot. The line you marked as "unretract" is not an unretraction. It's a normal extrusion move. The unretract move is two lines above (`G1 E5.00000 F4800.00000`).

lonelymyp, 2015-03-13:
Yes, you're right. 
looks like I'm not there looking for the problem.

When enabled the "wipe" retract slower due to the fact that it is synchronized with the movement of the head. Occurs several retracts and one unretract. looks like a different acceleration.

closed 2015-03-13 by lonelymyp