slic3r/Slic3r #2717: [Feature Request] Provide a way to disable bridge detection

Vicious-one, 2015-03-07:
Considering the growing number of reports and evidence on improper bridging in tricky situations, it would be good to have a way to disable bridge region detection for the entire model or within a modifier mesh.

lordofhyphens, 2017-03-27:
[image]

lordofhyphens, 2017-03-27:
[image]

closed 2017-03-27 by lordofhyphens

Vicious-one, 2017-03-27:
This option does not affect infill segmentation.

lordofhyphens, 2017-03-27:
How would disabling it help anything for infill?

Vicious-one, 2017-03-27:
It's fairly obvious.

[image]

[image]

Disabling bridge detection altogether can prevent the obvious anchoring failure.

alexrj, 2017-03-27:
@Vicious-one, it bridge detection fails the solution is to fix bridge detection. If you have a failing test case, please share it. We will be glad to work on it!

Vicious-one, 2017-03-27:
@alexrj, here it is.
Bridge detector grows infill regions in order to provide better anchoring, but this strategy is faulty in presence of thin walls. Solid unsegmented infill is clearly better in this case.

[image]
[image]

STL and config
[bridge-d1.zip](https://github.com/alexrj/Slic3r/files/872202/bridge-d1.zip)