slic3r/Slic3r #2719: wrong slicing calculation from stl file

vadimosk, 2015-03-08:
I adjust one gear from 5mm(5.6) to 4mm(4.6) inner diameter in FreeCad. I only add one 4mm tube into the center in the gear and then with another cylinder drill hole for 3mm bolt for tighten in to the shaft. In all programs (MeshLab, FreeCad, Wings, Cura, Slic3r/3D_preview) this part look OK. 
But when slicing in Slic3r (and also when is part printed) the gear hole looks  from bottom to top:
  0-~2mm -- 5mm diameter
~2-~7mm -- 4mm diameter
~7-   top  --- 5mm diameter.
The 4mm hole is only in the place where is the second hole for shaft bolt, the right is from bottom to top. I try more slicing options but not luck. Then i try Cuda and here is all OK. The hole is 4mm for whole height. 
I try Slicer 1.2.0, 1.2.6 and 1.1.7. The same. I try Cuda and here is slicing OK. But Slic3r is my favorite ;-)

The stl file here:
https://www.dropbox.com/s/vg89cpvwc8ctxr9/4mm_gear_maly_ISSUE.stl?dl=0

[image]
[image]
[image]
[image]
[image]

whosawhatsis, 2015-03-08:
I can see from the Wings view that it's not manifold. It looks like you started with the shape that Slic3r is creating when it slices and tried to add a tube to decrease the inner diameter, but you didn't perform a boolean union, so there are two separate solid volumes that are overlapping. This is impossible in physical space, so the second volume is being dropped.

Of course, I can't tell from the image if the added geometry even is a closed volume. If it's not manifold, you won't be able to do a boolean union, and there's no possible way for slic3r to figure out what you intended the part to look like.

vadimosk, 2015-03-08:
I must say i big amateur in 3D modeling. But when i see the straight hole 4mm in model i want print straight hole.  
Maybe help a FreeCad file?
https://www.dropbox.com/s/lldjoimu3p25uh0/4mm_Gear_3%20-maly_ISSUE.fcstd?dl=0

xoan, 2015-03-08:
As @whosawhatsis says, it's a model problem: in FreeCAD you need to convert meshes to shapes, perform a boolean operation (and optionally, refine the final shape) before to try to export it again.

vadimosk, 2015-03-08:
OK. Thanks. I try. As i said i amateur in 3D modeling.

closed 2016-07-13 by lordofhyphens