slic3r/Slic3r #2720: Feature request

imqqmi, 2015-03-08:
I own a Prusa i3 and adapted it to have Bowden setup. I've noticed when the speed of the extrusion just before a retraction/move is low the retraction is much more effective, no oozing or blobs on the next extrusion part. When the speed is high the pressure of the filament is still high and even with longer retractions the nozzle still oozes. This is not something I can control as sometimes an perimeter infill (not the zigzag infill but a single finishing extrusion inside the perimeter on narrow walls) does this at a very low speed like 15mm/s. 

Would it be possible to add an option to slow down by a given value the feed rate on the last extrusion just before it retracts and moves to a different part? This would effectively lower the pressure and prevents oozing in many cases I think. Retraction can then be lower which prevents many other problems like jams, under extrusion after retraction/resume etc. 

What do you think, would this be a good thing to have?

xoan, 2015-03-08:
Good idea, but I guess it could be done by using a post-processing script: detecting every retract, searching for a longer (than a predefined distance) previous movement and setting a lower feedrate before this movement.

imqqmi, 2015-03-08:
I was thinking about the same thing, can this be easily integrated in repetier host? Or does Slic3r offer a post processing option? It would be nice to have in a convenient GUI :)

xoan, 2015-03-08:
http://manual.slic3r.org/advanced/post-processing

imqqmi, 2015-03-08:
Thanks, I'll look into it!

jonaskuehling, 2015-03-09:
Sound like the "advance algorithm" feature is what you're looking for, read "active nozzle pressure control for acceleration and deceleration". You can either find it in your firmware (Marlin, Repetier..) or even in the latest Slic3r (see "Printer" tab > general > Advanced > "Pressure advance").

imqqmi, 2015-03-10:
@jonaskuehling I can't find the article you're referring to. I've tried pressure advance in the past. Even setting it to '1' it creates huge blobs of filament, which is visible in repetier host as well. 

Is there any documentation on how to calibrate this, could you post a link to it? It would be great if it works! 

I didn't know the Marlin firmware had such a feature, thanks for the heads up! Will look into that too.

jonaskuehling, 2015-03-10:
@imqqmi I must admit that I didn't try the advance feature that was recently built into Slic3r yet, since I prefer to do those kinds of hardware-related compensation in firmware rather than via GCODE. What I have experience with is the advance feature of Repetier firmware, which I think is based on the same algorithm like it is implemented on Marlin. That feature works quite well - has limits and can't solve all the pressure issues, but in general it does a fairly well job and in fact addresses exactly the over-/underextrusion problems during acceleration and deceleration.

Oh sorry, the "read [...]" recommendation rather ment "read it like [...]" or "that means [...]" - no reference to any particular article here ;-)

imqqmi, 2015-03-10:
Ok, thanks! I've tried the Marlin advance feature and after some research, it's broken and unusable. The extruder doesn't move at all when turning it on. So I'll abandon it for now. There's other firmware that has an implementation for this as well so I might try those if they are compatible with my printer/RAMPS/pololu board and the auto bed leveling system I have.

I'm now experimenting with slic3r advance feature. I'm now starting with 0.1 setting (I tried '1' before which may be way too high, hence the excessive extrusion). If this fails I'll try my own ideas of adapting gcode like in the first post. My feeling is (though it's not yet confirmed) that feed rate adjustments in the gcode could potentially work on any type of filament so that would cover most circumstances. I've noticed that the printer at a constant 15mm/s on the first layer has no blobs at all. So if I can get it to slow down to 15mm/s before retractions may solve a lot of problems. 

I agree that it would be much more convenient if the firmware solves this all on its own. I wonder if the printer is aware of the filament diameter and extrusion multiplyer through the gcode. The Marlin confifuration_adv.h has some defines for this, as well as extrusion steps per mm.

imqqmi, 2015-03-10:
Ok, I've done some testing with extrusion advance. The good thing is that it reduces the strings between parts but the downside is that upon resuming, after retract it extrudes a large blob. I've tried to adjust this with a negative 'Extra length on restart' but it seems that it will under extrude in places where no pressure regulation takes place. So the balance between pressurization and represurization is off and can't be adjusted. What would help is a variable to control the fewer extrusion and more extrusion separately. 
[image]
In the image above at the top you see the control without any extrusion advance. The second the extrusion advance is set to 0.35 and the third has EA set to 0.4 and extra length after retract set to -1mm. For blobbiness and strands this is the best one, though as you maybe can see there's some under extrusion on the right part in the upper right corner in the infill. 

The second one has a big blob in the left part in the upper right corner. This is where it starts printing after a retract. 

While the third one manages to reduce the blob, it also under extrudes in other parts.

lonelymyp, 2015-03-13:
such a function is in the Simplify3D. 
called "Coast at end". 
[image]

a4jp-com, 2015-03-14:
What does coast at end mean?

lonelymyp, 2015-03-14:
as you can see in the screenshot, "Turn off the extruder a short distance before the end of a loop to relieve pressure in nozzle and prevent blobs."
Filament pressure drops before retract

imqqmi, 2015-03-16:
Thanks for the heads up! I've also tried coasting in the post process script, and it helps a little bit. I used regular expressions and hit the 9 look back buffer limit and could only process 3-4 moves/extrudes. I'll rewrite it to be not dependent on regex so I can set the distance/number of moves without this limitation.

bubnikv, 2016-10-27:
I have implemented a prototype of volumetric extrusion rate change limiting into the Prusa3D fork. It is in an experimental state. It seems to be working somehow, but it breaks on the gap fills, where the volumetric extrusion rate changes rapidly. This is the same woe that the current Slic3r constant volumetric rate mode suffers.

This feature is very tricky to be implemented correctly. One has to optimize for multiple contradicting requirements: Minimize the print time, heed the acceleration profiles of the paths, equalize the nozzle pressure, respect the cooling profiles. One cannot just optimize one of them. They need to be optimized together.