slic3r/Slic3r #2721: [Bug] "Send to printer" fails with filenames containing special characters (e.g. umlauts)

GHPS, 2015-03-08:
The problem: The "send to printer" button in Slic3r 1.2.7 is a great and very useful feature. It works like charm except in cases where the filename contains special characters, e.g. German umlauts. While slicing (and saving) the G-Code is possible without any problem, the transfer fails with the error message

"Error while uploading to the OctoPrint server: 500 Server closed connection without sending any data back".

Workaround: Renaming the file fixes the problem.

Test case: Try to send "Test with Umlaut öäü.stl" to printer

Platform:
- Slic3r 1.2.7
- OctoPrint 1.1.1

lordofhyphens, 2015-03-08:
Can you upload the files with umlauts to octoprint already?

GHPS, 2015-03-09:
> Can you upload the files with umlauts to octoprint already?

Yes - uploads work fine, both 'drag-and-drop' and via the upload button. 
Although the file is renamed: 
1) Spaces are substituted by underscore.
2) Umlauts (öäü) are substituted by similar non-umlaut characters (oau).
3) The German s character ('ß') is stripped.

alexrj, 2015-06-02:
Fixed in 854be6a1861f109763d93974e8eef39db823ea8e

closed 2015-06-02 by alexrj