slic3r/Slic3r #2722: Bug 1.2.7-dev: First Layer height is wrongly applied to first layer on top of raft as well

jonaskuehling, 2015-03-09:
Hey,

in current 1.2.7-dev the first model layer on top of raft is generated with rather large layer height (0.43mm layer, while model material nozzle is Ø0.35mm), even though general layer height is set to 0.15mm.

I set first layer height to 0.5mm (Ø0.5mm support material nozzle) - which is now possible thanks to #2701 - and it seems like Slic3r tries to do it justice by also printing an as thick first model layer as somehow possible with a Ø0.35mm nozzle :-)

[image]

GCODE: https://www.dropbox.com/s/wi8wnfjx7y97ud8/cylinder20.gcode?dl=0

alexrj, 2015-03-09:
Fixed! And added a regression test.  I'll wait for your confirmation before closing this one. :)

whosawhatsis, 2015-03-09:
It's a little more complicated than that. The first layer height and extrusion width settings are used to get good platform adhesion, so the settings used for printing raftless are not also appropriate for use with a raft. OTOH, it's still useful to be able to tweak these settings to get the raft to release.

For easy reconfiguration (just activate/deactivate raft) the first layer settings should be applied to the first layer of the raft if there is one, otherwise to the first layer of the print, and there should be a separate set of the same settings within the raft settings to apply to the first layer printed on top of the raft. This is potentially confusing though, and using the first layer settings for the first layer of the object (so that they need to be changed when activating/deactivating raft) would be more easily understood, but more of a hassle to use.

The best solution I've found to this problem is in MatterControl. Rather than having different profiles that can be swapped-out, each machine has one printing profile, plus a set of modifiers that can contain settings from any section (print, filament and printer) and overrides only those settings. I have a modifier set up in that that to activate raft that also changes the first layer's height and extrusion width.

alexrj, 2015-03-09:
@whosawhatsis, I think there's a misunderstanding here as we actually seem to agree: either you missed the point of @jonaskuehling's post or I'm missing yours :)

The First Layer settings (extrusion width, height, speed, temperature etc.) are applied to the actual first layer that is laid onto the print bed. That happens to be the first raft layer if there's raft, otherwise it's the first object layer as usual. So I think we all agree on this.

@jonaskuehling was reporting that after my recent change (#2701) the first object layer got too thick even when printed above a raft, which is wrong. I have fixed that now. First object layer above raft honors the Support Material Contact Distance option, and ignores any First Layer setting.

whosawhatsis, 2015-03-09:
My point was that even when printing on a raft, it can still be useful to change the height/width for the first object layer (for raft release), and that avoiding confusion when making these options available requires more consideration.

jonaskuehling, 2015-03-10:
@alexrj I can confirm that the first object layer height on top of raft is now sliced with general layer height as intended. Tested a minute ago with latest master, thanks!

alexrj, 2015-03-10:
Okay @whosawhatsis, now I get your point. We have Support Material Contact Distance for that purpose, which applies to raft as well.

alexrj, 2015-03-28:
Can we close this? @whosawhatsis, are we all on the same track?

whosawhatsis, 2015-03-28:
Go ahead and close this one. My concern was slightly off-topic anyway, and if I encounter a real problem, I can always open a new one.

alexrj, 2015-03-28:
:-)

closed 2015-03-28 by alexrj