slic3r/Slic3r #2723: Request 1.2.7-dev: Use dynamic layer height for raft layers (= print raft layers as thick as possible)

jonaskuehling, 2015-03-09:
One more thing...

It would be good, if all raft layers would be printed at the maximal possible layer height (= support material nozzle diameter), just like Slic3r does dynamically for support material regions, where it only needs to build up height without supporting model regions.

Currently raft will be printed with rather thin layers, in case the general layer height is set to e.g. 0.15mm. Since the raft layers without doubt are entirely independet from model layers, it would form a much better raft (also faster, with less layers for the intended height..) when printed thicker.

Compare sample GCODE/picture of #2722!

alexrj, 2015-03-09:
Done!  Waiting for your feedback before closing this.

jonaskuehling, 2015-03-10:
Looks better now, at least in terms of raft layer height. But there seems to be another issue now, the topmost raft layer (support interface) seems to be generated kind of infinitely thin.

I configured 7 raft layers and 3 support interface layers.

[image]

[image]

Support interface layer below the object somewhere higher than in the raft region seem to be generated correctly with all 3 interface layers and correct layer heights.

jonaskuehling, 2015-03-10:
Sorry I forgot to attacht the sample GCODE: https://www.dropbox.com/s/hkyq03m2jwlgwqg/cylinder20.gcode?dl=0

jonaskuehling, 2015-03-10:
From the GCODE I see the following layer heights:

0.5 (raft Layer 1)
0.26 (raft layer 2)
0.26 (raft layer 2)
0.26 (raft layer 2)

simonkuehling, 2015-03-10:
Another example here - four problems in this case, maybe these help to locate the cause:
* 3rd raft layer thin (0.01mm)
* 2nd interface layer thin (0.01mm)
* 9 raft layers - although only 7 configured
* first object layer 0.24mm - instead of 0.25 like configured

```
z-pos   layer height 
0.35    0.35          (raft layer 1)
0.61    0.26          (raft layer 2)
0.62    0.01          (raft layer 3)  <-------------
0.88    0.26          (raft layer 4)
1.14    0.26          (raft layer 5)
1.40    0.26          (raft layer 6)
1.66    0.26          (raft layer 7 / interface)
1.67    0.01          (raft layer 8 / interface)  <-------------
1.93    0.26          (raft layer 9 / interface)
2.17    0.24          (object layer 1)
2.42    0.25          (object layer 2)
```

[2nd-config.ini](https://www.dropbox.com/s/itei6h8mz1r1r3s/2nd-config.ini?dl=1)
[2nd-model.stl](https://www.dropbox.com/s/sjwi91rlclhppet/2nd-model.stl?dl=1)
[2nd-paths.gcode](https://www.dropbox.com/s/9g2ax2eu5ez4p0b/2nd-paths.gcode?dl=1)

simonkuehling, 2015-05-06:
Hi Alessandro,

we are still thrilled like kids watching our ongoing tests of soluble support material with ABS+HIPS - this is going to be BIG news with your next Slic3r release, and we can hardly wait to publish our results to the community.

Unfortunately we are a bit stuck caused by the above issues as it renders the essential raft pretty much unusable at the moment (no offense to you on this!).

But - besides throwing money at you (our offer is still valid!) - is there anything else that could help to support you or relieve you of some work? Like throwing money at someone else for helping you (in case you know some developer that is already involved for example) - please let us know.
We are keen to help pushing forward some of these features urgently needed by the whole 3D printing open source community.

alexrj, 2015-06-02:
Hey @simonkuehling, this should be fixed now. Let me know whether we can close this one finally. :)

simonkuehling, 2015-06-02:
@alexrj, hmmm.. just gave it a try but no luck unfortunately:

```
z-pos    layer-height
0,5      0,5                        (raft 1)
0,51     0,01                       (raft 2)
0,88     0,37                       (raft 3)
1,25     0,37                       (raft 4)
1,62     0,37                       (raft 5)
1,62     0                          (raft 6)
2,0      0,38                       (raft 7)
2,0      0                          (interface 1)
2,38     0,38                       (interface 2)
2,75     0,37                       (interface 3)
3,0      0,25                       (object 1)

```

with identical config like above (7 raft layers)

alexrj, 2015-06-03:
Whoops, there was one more issue. I tested the wrong way.  Fixed now!

simonkuehling, 2015-06-03:
here we go! great work, @alexrj!
All my test cases are ok now...

alexrj, 2015-06-03:
:ok_hand:

closed 2015-06-03 by alexrj