slic3r/Slic3r #2724: "Cut -> Rotate lower part" places the part under the grid

gege2b, 2015-03-09:
Hi here
As mentionned in #2475 I got a strange behaviour while cutting an object

Cutting an object and keeping "rotate lower part" checked will make the lower part of the object to be rotated (thats good) but under the grid.

No matter what  the model or the values are, as long as "rotate lower part upwards" is checked

[image]

[image]

alexrj, 2015-03-09:
Oh no, again!

alexrj, 2015-03-09:
One line was missing. Fixed.

closed 2015-03-09 by alexrj

gege2b, 2015-03-10:
liked your "oh no again" :D