slic3r/Slic3r #2725: placement of models specified on command line does not honor bed dimensions

galexander1, 2015-03-10:
I made a simple stl with just a 1000mm rectangle in it.  That is much too big.  At 1.0.1, I would get this error:

1 parts won't fit in your print area!

But 1.1.7 and 1.2.6 and git newest all accept it without comment.  I don't know if the same problem exists in the GUI, though I suspect it does.

At 1.0.1, Geometry::arrange would get its $areax,$areay from Model::_arrange, which used to config->bed_size.  Now there is a $bb, passed down all the way from Model::duplicate and duplicate_objects.  But the calls to those two subs from Print::Simple::set_model doesn't provide a $bb.

I'm going to attempt to get the info from config->bed_shape in Print::Simple::set_model, and submit a pull request if successful... But I figure whoever changed this probably has a grand plan and I'd rather not get in the way of that..

Thanks! I love slic3r as a user, and now that I'm looking under the hood, I am even more impressed!!

galexander1, 2015-03-15:
Well, I made the change in pull request #2736.  It works for me, in that now it complains if I use commandline to add an object that is too big, and it centers the object in the center of the bed_shape region.  I'm not sure if this is really what the plan was when bed_shape was introduced, so I sure won't be offended if a different patch is prefered.  But for me this patch was necessary in order to be able to use the new Slic3r, otherwise it tried to place the object centered at 100,100 no matter what.

BTW - new Slic3r (version 1.2.x) is *MUCH* faster than version 1.0.1.  Thanks!!!!

alexrj, 2015-03-15:
Let's first start from identifying what problem(s) are you reporting. Good description of problems is the necessary premise before jumping to solutions.
As far as I understand you're reporting three distinct things:

1. Command-line slicing will not prevent you from supplying a 1000mm model even though your print bed is supposed to be smaller.
2. Command-line slicing will center objects objects at 100,100 by default instead of computing the center point dynamically from the configured bed shape.
3. Command-line slicing will not take the actual bed shape into account when auto arranging parts.

Regarding point 1, that happens by design. Command-line slicing does not enforce bed size for validating model input because we just do what user wants. the CLI API is used by host software (for example Repetier-Host) that already check print boundaries, and we don't want to require additional data from them. Still, a warning and/or an additional command line option to enforce bed size could be useful. But it shouldn't be based on the `arrange()` algorithm (which can lead to false negatives) anyway.

Regarding point 2, there's a `--print-center` option that allows to set the center point by overriding the default 100,100 coordinate. I agree that the default value (that is, when `--print-center` is not supplied) could be calculated dynamically as the center point of the bed shape.

Regarding point 3, I think the bed shape should be passed to `arrage()` as a pure hint, but it shouldn't enforce any kind of validation.

No need to update your patch, I'll do these changes.

galexander1, 2015-03-15:
Thanks!  Sorry my patch only took into account my one usage scenario.

lordofhyphens, 2016-12-31:
@alexrj is this closable then?

alexrj, 2017-03-13:
All done. These changes make a lot of sense. Thank you @galexander1 for suggesting them.

xoan, 2017-03-27:
This is again reproducible in current master :(

alexrj, 2017-03-27:
@xoan, what exactly? Don't you get a warning?

xoan, 2017-03-27:
If I slice a model through command line without passing `--print-center` option, using a printer profile with a 140x140 bed, print is centered at 100x100.

`slic3r --load config.ini file.stl`

[config.ini.gz](https://github.com/alexrj/Slic3r/files/873554/config.ini.gz)

xoan, 2017-03-27:
Oh, wait... The option that is ignored is `--load`, so it is a different unrelated issue.

xoan, 2017-03-28:
I guess this can be closed again because #3816 

Sorry.

closed 2017-03-28 by alexrj