slic3r/Slic3r #2726: 1.2.7-dev Support regression (?)

lordofhyphens, 2015-03-10:
I've noticed that slic3r 1.2.7-dev likes to put support structures in areas that are more like a bridge (like when the area is supported on two sides or even three). Thus, things like nut traps get filled in(!) and the support material is also very difficult to remove from these tiny areas. Is it possible to make the support detection less aggressive?

Some further checks with http://www.thingiverse.com/thing:12925 indicate that "Don't Support Bridges" doesn't actually do anything, as Slic3r will still try to support the bridge.  

[image]

Screenshot is from https://github.com/lordofhyphens/Prusai3_EINSTEIN_Reworked/blob/master/Pieces%20en%20STL/EXTRUDEUR/BODY-EXTRUDEUR-WADE/BODY-EXTRUDEUR-WADE.stl with support turned on, but you can make this trigger on a 5mm step cube.

I went back and checked, and the same config generates the same problematic support structures in 1.2.0. 

OS: Debian Linux x86_64 (Jessie).  
Config: 
# generated by Slic3r 1.2.7-dev on Mon Mar  9 23:35:10 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,190x0,190x220,0x220
bed_temperature = 110
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 0.9
bridge_speed = 50
brim_width = 6
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = G10; quickly retract\nM104 S0\nG28 X0  ; home X axis\nM84     ; disable mots\nM140 S0\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 0.96
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 2.85
fill_angle = 45
fill_density = 15%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 110
first_layer_extrusion_width = 1
first_layer_height = 0.26
first_layer_speed = 25
first_layer_temperature = 235
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 1
infill_overlap = 20%
infill_speed = 50
interface_shells = 0
layer_gcode = 
layer_height = 0.24
max_fan_speed = 100
min_fan_speed = 100
min_print_speed = 20
min_skirt_length = 5
notes = 
nozzle_diameter = 0.5
octoprint_apikey = 543BC61C6D3240F78130EA181149B6F6
octoprint_host = 192.168.1.104:5000
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].[print_preset].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 50
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 4
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 11
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 18
small_perimeter_speed = 50
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 50
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = M106 S0\nG28 Y0 Z0 X0 ; home all axes, \nG90; set to absolute position\nG1 Z0.1; leave some room for oozing\nM190 S[first_layer_bed_temperature];\nG28 Y0; rehome y\nG1 Z10\nG28 Z0 ; make sure we are homed\nM140 S[first_layer_bed_temperature]; Wait until heatbed hits [first_layer_bed_temperature] \nM109 S[first_layer_temperature]; heat nozzle to [first_layer_temperature]C;\n; wait 2m?\nG11 ; end retract
support_material = 1
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 1
support_material_interface_spacing = 0.05
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 50
support_material_threshold = 0
temperature = 230
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 50
use_firmware_retraction = 1
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = -0.25
z_offset = 0

MikeFitz73, 2015-03-15:
yeah I'm having these issues with 1.2.6.  even more, say you have a object that is the same on left and right side.  bridge areas on left side get supported and identical bridge areas on other side do not get supported.... when there should be no support for either of these areas.

I would like to be able to delete support manually for certain areas.

lordofhyphens, 2017-03-19:
[image]
Looks like it has to do with some of the toolpath generation, if a perimeter doesn't form a full loop (which can happen in some instances) support gets made for it.

Test case (for convenience): [2726.zip](https://github.com/alexrj/Slic3r/files/852883/2726.zip)

Last checked on 90395f93940410093244e9cfcaf3dce407ddc811