slic3r/Slic3r #2727: Sequential printing - bed and nozzle temperature settings ignored after the first object is printed

zsolta, 2015-03-10:
I put a few objects on the plate and printed them sequentially. The temperature settings were
Extruder : First layer temperature = 190; Other layers =195
Bed: First layer = 60; Other layers = 50

The generated G-code has the right temperatures set for the first printed object only,  the following objects have only the Bed First Layer Temp set in the G-code, all other three temperature settings are missing. This looks like a bug to me.

Thank you.

Windows7; Slicer 1.1.7

lordofhyphens, 2015-03-10:
There are guidelines for posting bug reports, please review them.

This is also probably the same problem as #2702   but without a full, reproducible test case it's hard to tell.

alexrj, 2015-03-22:
Yes, looks fixed.

closed 2015-03-22 by alexrj