slic3r/Slic3r #2728: Bug 1.2.7-dev (Ubuntu 14.10): 3D view disappears during sliding/zooming

jonaskuehling, 2015-03-10:
Hey,

thought I read about something similar here lately but can't find it again at the moment. When moving/zooming the viewpoint in 3D view (plater as well as path preview) the rendered parts suddenly disappear sometimes. I think it mostly happens when the viewpoint is near to front view/side view or going below the build plate.

Hard to describe or illustrate actually....

Let me know what kind of details you need to reproduce the issues, I'll deliver :)

a4jp-com, 2015-03-10:
That was my post. It looks like the bottom of the screen has been fixed just the top has a problem for me now. I'll post a picture later on tonight.

VanessaE, 2016-06-28:
Still a problem with current git.  Also related to this, panning gets screwed up after you're zoomed in and have rotated once or twice already - trying to pan up, for example, moves the viewport to the left, as if the camera is trying to move along the lines on the floor grid.

alexrj, 2017-03-20:
In my experience, this happens with almost all 3D visualization software. :-)))
We now have menu items for restoring the zoom and center, I think this is enough.

If you strongly disagree, feel free to reopen (suggesting a solution) :-)

closed 2017-03-20 by alexrj

a4jp-com, 2017-03-20:
I think that was the bug I was talking about before where the objects and platter would get clipped out. Maybe this problem has been fixed. If items just disappear in programs you use there might be a bigger problem. Slic3r is working pretty well now. I'm using the newest version of the complied software though.