slic3r/Slic3r #2729: Bug 1.2.7-dev: "Can't call method "get_field" on an undefined value [...]/lib/Slic3r/GUI/Tab.pm line 414.

simonkuehling, 2015-03-10:
Just noticed a crash of 1.2.7-dev @13b63d06edceeea22d56d797888985b7551304fa while changing some parameters in _Print Settings -> Support Material_ and got this error:

```
Can't call method "get_field" on an undefined value at [...]/lib/Slic3r/GUI/Tab.pm line 414.
```

Unfortunately i cannot reproduce this yet, not sure which values were srewed there - but maybe an exception catch or the like is all that is needed here..?

Ubuntu 14.04 64bit

alexrj, 2015-03-23:
@simonkuehling, where you able to reproduce this?

simonkuehling, 2015-03-23:
Unfortunately not - i'll keep an eye on that in case the crash reoccurs...

I suggest we should close this issue for the moment...

alexrj, 2015-06-02:
I think we can close this as it's a duplicate of #2729 and I fixed that one

closed 2015-06-02 by alexrj