slic3r/Slic3r #2731: Crash on loading stl: perl': realloc(): invalid old size

TTN-, 2015-03-11:
```shellscript
user@pc:~/Scripts/Slic3r-dev$ ./slic3r.pl 

(slic3r.pl:3392): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17

(slic3r.pl:3392): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17

(slic3r.pl:3392): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 17
*** Error in `perl': realloc(): invalid old size: 0x00000000077354e0 ***
Aborted
user@pc:~/Scripts/Slic3r-dev$ ./slic3r.pl
```

Using 13b63d06edceeea22d56d797888985b7551304fa
Ubuntu 14.04 x64
perl v5.18.2

Slic3r crashes on loading a huge stl of the scara helium frog: http://www.heliumfrog.com/hf06robot/Helium%20Frog%20Scara%20HF06.zip
Crash occurs with HF06 Scara Assembly 05.stl

I accidentally loaded this part instead of another, but I thought this crash info may be helpful somehow, or may occur with other stl's out in the wild. If not please close the issue. Less clutter is better :)

alexrj, 2015-03-28:
Well, thank you for the report. I can't reproduce any crash on OS X. That STL file is broken though. I'll test on Linux.

fgelm01, 2016-09-10:
Saw this as well. I'm on Ubuntu 16.04x64 and ran into this problem after leaving the window open for a while and not doing anything. The model I had loaded was not large or complicated, but I did have the 3D preview showing.