slic3r/Slic3r #2732: Feature request: disable cooling on bridges narrower than ...X... mm or layer time > ...X... sec

karrack, 2015-03-11:
i noticed that when slicing objects with infill and activated cooling option, when printing the fist solid layer of the top solid layers, slicer would assume it is bridging (which is correct IMO) and therefore use bridging speed and bridging cooling parameters. However this could lead to problems because some big and flat solid top surfaces i printed led to too much cool air getting to the nozzle, causing filament jams due to low temperature or even cold extrsuion (which was prevented by firmware)

However, cooling on these layers is often not needed as bridges are narrow and print time is high.

is there a way to implement a feature that would allow users to tweak the cooling settings a little more, such as: Disable cooling on bridges narrower than ...X...mm? or disable cooling on bridges when layer print time exceeds ...X... sec?

xoan, 2015-03-11:
Duplicated: #2598

lordofhyphens, 2016-07-11:
See if modifier meshes help.