slic3r/Slic3r #2733: Alternating direction for perimeters.

Android78, 2015-03-12:
There doesn't appear to be any option to alternate the direction for perimeters.  
This would be especially useful when trying to print very low layer height.  Tricky areas with low layer height (sloped slide with sharp corners especially), there is a tendency for the head to drag the previous layer which can cause tears in the side of the model if using low or no infill.  By alternating the direction (every even layer clock-wise and odd anticlockwise for example), the plastic that had been dragged on the previous layer would be dragged back on the next.

rsatue, 2015-03-12:
We have noticed the same issue. 

When printing close to the minimum overhang angle, it is always the same corner that tends to curl and it seems to be due to the perimeter printing direction. 

Alternating the direction of the perimeter print would probably compensate the effect of the direction and improve the finishing.

alexrj, 2015-03-12:
Thank you, duplicate of #1073

closed 2015-03-12 by alexrj