slic3r/Slic3r #2734: Building and testing Wx-0.9926 ... FAIL  Ubuntu 14.10 64 bit

megalomaniac, 2015-03-12:
Building and testing Wx-0.9926 ... FAIL
! Installing Wx failed. See /home/bogan/.cpanm/work/1426162407.7396/build.log for details. Retry with --force to force install it.
Wx::GLCanvas is up to date. (0.09)
The following prerequisites failed to install: Wx

HOW DO I FIX THIS?? 

Also: 
Configuring Net-Bonjour-0.96 ... OK
Building and testing Net-Bonjour-0.96 ... FAIL
! Installing Net::Bonjour failed. See /home/bogan/.cpanm/work/1426162403.7312/build.log for details. Retry with --force to force install it.
Don't worry, this module is optional.

To fix this do I just sudo apt-get install Net-Bonjour-0.96  ???

Tuxomx, 2015-03-15:
Hi. I got the same problem (the Wx one). I have no idea why this happens, but checking build.log I found a nonexistent routine. Searching on internet I found somewhere to use SetSelection(-1) instead of RemoveSelection, so I did that on file Controls.c and it just compiled.

I am currently using fedora 20, and what I did and worked for me is this:

1: change to build directory (in your case /home/bogan/.cpanm/work/1426162403.7312)
2: Enter Wx-0.9926 directory
3: Edit  Controls.c and search for 
       THIS->RemoveSelection(); 
   and change to
       THIS->SetSelection(-1);
   (there are two of these ones)
4: make
5: make install
6: go back to your slic3r source and compile again.

Regarding Net-Bonjour I have no idea, but you can just check the log and search for the error, but if you have Net-Bonjour as a package I suggest you to install it.

Must say this is not a Slic3r problem, and this developement version is working quite ok.

SALUDOS
ALVARO

lordofhyphens, 2015-03-15:
If you're using Ubuntu, you can pull my branch instead, which I use to compile debian packages and keep in sync with the master branch here.
Make sure you've got the packages needed to make pkgs (see ubuntu docs).

https://github.com/lordofhyphens/Slic3r/tree/debian

from the dir: fakeroot debian/rules binary

This should make a .deb file that you can then install. If the package complains about missing libraries, then install them.

lordofhyphens, 2016-07-12:
@megalomaniac going to close this for now, as issues with Wx isn't Slic3r's fault and you never updated this topic with any success and/or failure.

closed 2016-07-12 by lordofhyphens