slic3r/Slic3r #2735: feature request. Wipe after retract.

lonelymyp, 2015-03-13:
Is it possible to add features "Wipe after retract"?

Wipe is synchronous with retract, retract divided into several short pieces until the nozzle moves.
At low speed printing, with Bowden, retract going slowly and leak visible.
if first came full retract, and then wipe, leak would be less.

alexrj, 2015-12-19:
"Low speed printing" does not affect the speed of the wipe-while-retract moves. Extruder motor is reversed at the same speed as it would be in your proposed scenario (wipe after retract). Thus, the leak which is currently distributed along a path would be dropped at a single point making a blob...

lonelymyp, 2015-12-20:
extruder motor moves quickly(50mm/s), but the retract is interrupted by pauses while moving the nozzle(20mm/s).
total retract time = retract time + time movement of the nozzle

as it is now :
- first short part of retract, the pressure in the nozzle remains
- movement of the nozzle and the leak appears.
- second short par of retract, the pressure in the nozzle remains
- movement of the nozzle and the leak appears.
- final short par of retract, no pressure.

how to do it:
- full retract, no pressure in the nozzle remains
- movement of the nozzle, wiping
- movement of the nozzle, wiping.

ghuser2, 2016-01-02:
Wipe currently occurs at the same time as retract? I though the retract was first, followed by wipe.

lordofhyphens, 2018-11-18:
@ghuser2 the code should be doing the move on both axes.

lordofhyphens, 2018-11-18:
@lonelymyp it occurred to me that a version of what you described would allow for some wiping to occur during firmware retracts, assuming Z lift is delegated to Slic3r. 

That is:
* move into infill
* perform the retract 
* move to position
* unretract

supermerill, 2018-11-20:
Isn't this issue a bit old?

In an ideal world, the firmware already remove any pressure in the nozzle thanks to "pressure advance" (or other feature with the same aim).

The retract is to try to delay oozing enough, to have time to do the move? 
and the wipe to remove the string?

The concern may be valid, and maybe we can cut the retract& wipe in two phases: 
* wipe&retract: short enough to be at full retract speed as set in the extruder param. Slow down the wipe if the extruder is too slow.
* continue wiping if needed