slic3r/Slic3r #2741: Crossing perimeters although it shouldn't

lahure-c, 2015-03-16:
Hi
I would like to avoid crossing perimeters in order to avoid oozing deposit on visible surfaces
But I cannot succeed.
Here is the gcode file
https://github.com/lahure-c/public/blob/master/CylAR-4.gcode
; generated by Slic3r 1.2.6 on 2015-03-16 at 11:42:42
;...
; avoid_crossing_perimeters = 1

When completing the infill, the head moves across the internal perimeter instead of choosing the  longer path of flying above the deposited infill.
That's what I see with this gcode viewer http://gcode.ws

Thank you

lahure-c, 2015-03-16:
Here's a snapshot of the crossing:
https://github.com/lahure-c/public/blob/master/CylAR-4.jpg

lahure-c, 2015-03-16:
I have found a fix to this problem
Although I have set the retract_length to zero, it seems that Slic3r still uses the retract-before_travel variable.
There retract-before_travel was set to 20.
Now that I set it to 2, there's no crossing anymore.

lahure-c, 2015-03-16:
I still have issues on a more complex model. 
It works fine with extrusion widths equal to 0.5mm for perimeters and infills.
But with other values it does not work, for example 0.4mm
May I post this ?

alexrj, 2015-03-28:
Have you read the guidelines? You probably haven't (why?). This report is incomplete.

lahure-c, 2015-03-30:
https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

You can close this thread.
I'll open a new one that will include new things and follow the guidelines

closed 2015-03-30 by lahure-c