slic3r/Slic3r #2746: German sharp s (ß) leads to error "preset does not exist anymore"

larsen0815, 2015-03-18:
I opened the slic3r configuration via RepetierHost and saved my print settings as "testß".

This led to "The selected preset does not exist anymore (c:\users\myuser\AppData\Roaming\Slic3r/print/testß.ini).", followed by 'I was unable to load the selected config file: Can't call method "has" on an undefined value at c:/progra~1/Repetier-Host/Slic3r_1.2.x/lib/std/Slic3r/GUI/Tab.pm line 236.', and the first error message again.

The profile still appeared in the dropdown list, but selecting it led to the same errors again.

slic3r 1.2.6
RepHost 1.0.6
Win 7 SP1, 64-bit

alexrj, 2015-03-21:
Thank you, this was fixed in 1.2.7-dev.

closed 2015-03-21 by alexrj

larsen0815, 2015-03-21:
Thx. I forgot to test if umlauts (äöü) are a problem, too, but I assume these would be fixed, wouldn´t they?