slic3r/Slic3r #2749: Gaps using washable Supports

PRO250, 2015-03-19:
I have a few questions:

1. When generating spherical or quickly converging surfaces (when the number of perimeters > 1) I often see holes in the perimeters. 
2. When printing models using soluble support is often necessary to adjust the gaps between the support and the model by XY. It would be great if this gap can be regulated. 
3. When generating support and “Contact Z distance” parameter set  in 0   the interface layers are not dense very often.
4. Maybe you will add the ability to print an individual column when using two or more extruders to equalize the pressure? 
5. In order to improve the quality of the surfaces adjacent to the soluble support, could you add the perimeters to the layers of interface?

Help me, please

lordofhyphens, 2015-03-19:
Need pics, copy of a model you are seeing the problem with and the slic3r
configuration you are using.

PRO250, 2015-03-19:
photos and G-code are attached:
https://www.dropbox.com/s/6u9n7imcpsf38zl/IMG_0803.JPG?dl=0
https://www.dropbox.com/s/h3tgd4wkul1e2tk/1.jpg?dl=0
https://www.dropbox.com/s/iy23yf5wov8e1zp/2_3.jpg?dl=0
https://www.dropbox.com/s/nrk8rj16ee4eqs6/composition.gcode?dl=0

lordofhyphens, 2015-03-19:
Thank you.

Is an export of the configuration you used included?

angar79, 2015-03-20:
You can take the configuration from g-code file also.
And about the questions from the topicstarter - I have the same problems too.

bethebeer, 2016-12-07:
I can add a picture to this. Just using a sphere with 23mm diameter and generating support with an 85 degree threshold. I have tried many different combinations of settings. I have attached the stl file as a txt file, and a screenshot of just the support material. This should be easily reproducible. I have used 1.2.9 and 1.31.6-prusa3d.  

[image]

[Untitled.txt](https://github.com/alexrj/Slic3r/files/635419/Untitled.txt)

bubnikv, 2016-12-07:
If you set the number of interface layers to zero, the "interface loops"
will disappear. We will likely have a release, where you could disable the
interface loops.

other than that, you would make me happy, if you could describe your issue
with more detail.

bethebeer, 2016-12-07:
Thanks bubnikv! That fixed it. Still not sure why it only creates those bumpy interface layers on a portion of the support, but it's looking good now.

bethebeer, 2016-12-08:
@bubnikv , now I am having a different issue. It seems that there is only one perimeter generated for the support material. This causes the support material to sag significantly. So there is not a nice surface to print on. I am using soluble support. Is there a setting for this or should I open this as a separate issue? Thanks again.
[image]

lordofhyphens, 2016-12-08:
Reduce the pattern spacing. The point of the interface, by the way, is to
produce that nice surface to print over, so you could put that back and
decrease the spacing for that.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bethebeer, 2016-12-08:
@lordofhyphens for some reason though, the interface layers are always appearing jagged, even when the spacing is down to 0. At 0 spacing, the interface layer doesn't seem to run parallel to the supported object as one might expect. The support also changes distance from the object. I have attached two pictures of the preview in slic3r that are only one layer apart. The pictures are of a spacing of 0.1, but they look nearly identical to a spacing of 0.

[image]
[image]

bubnikv, 2016-12-08:
> <https://cloud.githubusercontent.com/assets/1959490/20993650/48eeba94-bcb1-11e6-9860-11bf4f555058.png> [image:
> screen shot 2016-12-07 at 7 12 36 pm]

First, please try both the @alexrj and prusa3d builds. I may have
introduced some issues in the 1.31.6 build regarding the gaps between the
supports and the object, namely the supports may be half a perimeter shrank
compared to the @alexrj build. Actually I saw @alexrj pulled the C++ infill
changes from the prusa3d fork, so the newest Slic3r may be affected as well.

Second, to the interface loops and soluble interface. Currently it is not
possible to have separate main / interface support layers without having
the interface loops. You either have to disable the interface layers and
make the base support dense, or live with the interface loops.
We at prusa research are working on multi material printing with single
extruder and filament switching. For that system to work, I made a
prototype to synchronize support layers with object layers, and to disable
the interface loops.
https://github.com/prusa3d/Slic3r/tree/supports-synchronize-layers

It will one day come into the official prusa3d builds.

Why there are no interface layers along the sides of the sphere? Because
the sphere supports itself there. We may argue, that the soluble interface
shall be printed on the sloping sides. Maybe one day they will, but not
with the current support code base.

@lordofhyphens I am working on the new C++ supports, they are completely
revamped, therefore I am not inclined to spend time over the old supports.
I have found out, that the old Perl support code does not work on complex
large objects at all due to the limitations of the clipper library when
processing many tiny complex polygons. The support generation simply never
finishes, even after couple of days. There are two solutions to that:
Either to simplify the contours on the contour level by filling in the
concave kinks (the Cura way) or to cast the contours into a grid and
enclose the support towers into grid cells (the Simplify3D way). I have
tried to prototype both and the 2nd solution is much easier to implement
because I already had the EdgeGrid on hand and it is very fast and robust
indeed. The contour simplification on the other hand is much more difficult
to do right and requires much more tinkering and tuining and it will be
algorithmically much more complex and less robust. I think there is yet a
better solution, to start with the grid and then to shrink wrap a contour
around the content of the grid cells. I will try that one day, but it does
not have a priority as of today.

By the way, the easiness of debugging the C++ code in Visual Studio makes a
day / night difference over fighting with the Perl, especially when errors
in invocation of the XS code from Perl leads to crashes, which could only
be discovered by selectively commenting out pieces of Perl code.

lordofhyphens, 2016-12-08:
> bcb1-11e6-9860-11bf4f555058.png> [image:

@bubnikv agreed on moving to C++, I saw the work in your fork. It probably
will be need work on our end (porting to use boost instead of c++11 to
build on older compilers because that's what @alexrj uses).

bubnikv, 2016-12-08:
> @bubnikv agreed on moving to C++, I saw the work in your fork. It probably
will be need work on our end (porting to use boost instead of c++11 to
build on older compilers because that's what @alexrj uses).

What compilers? Do we really need to maintain the legacy? Look at the new
cura. They use C++14.

All compilers on our build system (Strawberries, not a newest Debian and
OSX Maveric) support C++11. I decided to go with C++11 in the Prusa3D
build, I don't see a point of supporting older compilers. Let's look into
the future.

lordofhyphens, 2016-12-08:
I am with you on the c++ standards!