slic3r/Slic3r #2751: Dimensional accuracy 1.2.6 experimental

CircuitGrove, 2015-03-20:
First, I'm loving 1.2.6.  I found a bug.

I have a printer with a 0.4mm nozzle.  I'm printing this calibration block:
[image]

Each step is 10mm thicker.  With 1.2.6, they all measure 0.2mm thicker than they should and the inside cubes measure 0.2mm less.  I get good dimensional accuracy printing with the same settings with 1.1.7.

lordofhyphens, 2015-03-20:
Please post a link to the model and export your config and post it. It
helps if we can replicate your situation.

CircuitGrove, 2015-03-21:
This is where you can find the model:  http://www.circuitgrove.com/resources/circuit-grove-accuracy-check-calibration-cubes

alexrj, 2015-03-21:
Not a bug, see http://manual.slic3r.org/troubleshooting/dimension-errors

closed 2015-03-21 by alexrj

CircuitGrove, 2015-03-21:
Hi Alex,

I apologies, perhaps it's not a bug, but it's not my machine either.  I can go point by point down that list and explain why it's not my machine.  Can you elaborate?

It prints great in Cura and it prints great in 1.1.7.  I use the same setting in 1.1.7 as I did in 1.2.6.  In 1.2.6, the outer dimensions are larger by 0.2mm, always, and the inner dimensions are smaller by 0.2mm.

My guess is on your side, you've sliced it in 1.2.6 and printed it out and it came out great?  Then I must be doing something wrong...

Can you perhaps help me understand what the problem is so that I also know that it's not a bug?

alexrj, 2015-03-21:
Nobody said it's your machine. The page I linked lists all the possible reasons for dimensional issues and recommended workarounds. If you think the error is in the G-code, prove it and show us what G-code lines are wrong...

a4jp-com, 2015-03-22:
The G-code looks okay from here but I think CircuitGrove has a good point.

An XYZ scale option would be very useful especially when trying to fit parts together. I know most people say it is best to muck around with settings and get the model right but at times we just want a quick fix. An XYZ scale option would be amazingly useful.

For example, if I'm making a round peg to fit in a hole but I want it to fit loosely in one area but firmly in another the XYZ scale settings would easily allow us to adjust the tolerance. Real holes are never perfect sizes when drilled by hand.

This XYZ scale option would also help people like me, that bought a machine but the warranty says I can't open the machine to adjust the settings or tune the hardware. So as it is impossible to adjust the settings, without voiding the warranty. I even sent it back to the maker to get it adjusted but they didn't do it properly. A setting like XYZ scale would help me change the X scale by a tiny bit to get perfect squares/cubes. Please consider adding this really useful setting. Even though I know you like fixing problems rather than hearing solutions first, as you said before.

I'm really sure these settings would fix CircuitGrove's problems. Even though it just looks like CircuitGrove needs to adjust the 3D printer's settings or hardware.

alexrj, 2015-03-22:
We already have an option named *XY Size Compensation*. Error is absolute, not relative. Scaling wouldn't help at all.

CircuitGrove, 2015-03-22:
Hi Alex, 

I'm comparing G-Codes now.  The tool path is identical between 1.1.7 and 1.2.6.  I'll keep digging to try and identify my differences.  I'll update when I have an answer.  

Cheers, and thanks.

André

CircuitGrove, 2015-03-22:
So I've been experimenting with a 20mm box.  I've generated gcode that prints a single perimeter only, no top/bottom, infill, brim etc...

I've found that the toolpath for both 1.2.6 and 1.1.7 is 19.6mm thick.  However the extrusion amount is ~0.65mm in 1.1.7 and ~1.0mm in 1.2.6.  I must have a setting discrepancy right?

http://www.circuitgrove.com/sites/default/files/Resource_Files/20mmbox-1.1.7.gcode
http://www.circuitgrove.com/sites/default/files/Resource_Files/20mmbox1.2.6.gcode

a4jp-com, 2015-03-22:
Is it possible to get separate XY Compensate values? The hand drilled holes are sometimes a bit elliptical.

alexrj, 2015-03-22:
@CircuitGrove, you seem to be missing the point that dimensional accuracy depends on several physical factors that are described in the page I linked above. Unless you can prove that those factors are not affecting your environment, I invite you to use the XY Size Compensation option whose goal is exactly the one you're looking for.

@a4jp-com, no, it wouldn't make sense. Your printer has backlash. Fix it, tighten belts.

CircuitGrove, 2015-03-22:
OK.  I will go point by point down the list in hopes that I can provide satisfactory proof.

-Vertical dimensions:  not applicable here, it's not what we are discussing.
-Plastic shrinkage: I'm using identical settings and plastic (PLA with heated bed) with 1.1.7 where there are no dimensional accuracy problems.  Besides, shrinkage would make both holes and perimeters smaller.  I'm seeing larger perimeters.
-More material is deposited in the inside:  Would cause a reduction in outside perimeters, plus I'm printing squares so this doesn't apply here.  It's all straight lines.
-Curves are approximated by polygons:  Not applicable, I'm printing squares
-Filament tends to cut corners:  Not applicable, I'm printing squares and for outside perimeters this would lead to dimension reduction.
-Z wobble:  This is a printer problem, Since I'm printing perfectly fine in 1.1.7, we can scratch this one off the list too.
-Non-regular filament section:  Same as Z wobble.
-Flow Math:  may be a source of my issue considering the tool paths appear identical between 1.1.7 and 1.2.6.

alexrj, 2015-03-22:
Use XY Size Compensation. I can't stress that more. If an older version produced larger objects that might work for you but of course didn't work much for others, so that's not good logic.

Anyway, use XY Size Compensation. That's why we have it. Please.

CircuitGrove, 2015-03-22:
Ok, I will.  Thanks.

zsolta, 2015-03-23:
Is there anywhere described what XY Size Compensation means? Is it like scaling the object in the XY plane only, with Z unchanged? The tool tip is not too verbose. Thanks

alexrj, 2015-03-23:
@zsolta, see the manual page I linked above.   Since the name is XY, it does not affect Z.

whosawhatsis, 2015-03-23:
@zsolta it doesn't scale the object, it offsets the perimeters to allow you to adjust tolerances. When the outer dimensions get smaller, inner dimensions (holes) get bigger.

zsolta, 2015-03-23:
All clear now. Thanks for all answers!

CircuitGrove, 2015-03-23:
FYI, adjusting this setting solved my problem.

derekhe, 2015-05-04:
@zsolta @alexrj 
XY compensation is quite useful to adjust the hole. But the problem is the outer dimensions are quite good for my print, only the inner dimensions get too small. If I adjust the compensation, the inner dimensions can be good but in the meantime the outer dimensions become smaller which is not expected.
Could we have a feature to set different compensation for inner parts and out parts?

ronindev, 2016-01-03:
@derekhe +1
I have perfect outer dimension. But inner holes are small. After calibrating inner dimensions with XY Compensations inner dimensions are ok but outer dimensions smaller now

a4jp-com, 2016-01-03:
I have reloaded my firmware and can get very accurate prints but it sounds like you want to do something really difficult. My idea for having separate X, Y and Z scale values wouldn't fix this type of problem. You might just have to remodel the object in 3D or punch out/drill a larger hole. The only other thing I can think of is adding a modifier with a scaling factor but I don't think there is a subtract area modifier. I think it would be nice to make a different thread asking for a modifier that can be used to remove areas. 

Object > Settings > green + icon