slic3r/Slic3r #2753: bug: version 1.2.6 Mac - crash when slicing

EricBabin, 2015-03-23:
Slic3r crashes when trying to slice the STL file from Thingiverse: http://www.thingiverse.com/thing:492101/#files

Load it into Slic3r. Rotate -90deg on X axis to lay it flat on it's bottom side.

Will paste the config file in separate comment.
This is my first posted issue. Please let me know if you need any further information.

Thanks,

Eric

EricBabin, 2015-03-23:
And the settings ini file....

# generated by Slic3r 1.2.6 on Mon Mar 23 16:52:34 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,298x0,298x275,0x275
bed_temperature = 60
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 1000
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 72
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 1500
disable_fan_first_layers = 2
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; Turn off extruder temperature\nM140 S0 ; Turn off bed temperature\nG92 E0 ; Reset extrusion distance\nG1 X0 Y190 F9000 ; Home X axis, Move the bed out, Limit to XY max\nM84 ; Disable motors
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 45
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 0.93
extrusion_width = 0.42
fan_always_on = 1
fan_below_layer_time = 60
filament_diameter = 2.85
fill_angle = 45
fill_density = 30%
fill_pattern = rectilinear
first_layer_acceleration = 600
first_layer_bed_temperature = 60
first_layer_extrusion_width = 0.53
first_layer_height = 0.35
first_layer_speed = 75%
first_layer_temperature = 210
gap_fill_speed = 55
gcode_arcs = 0
gcode_comments = 1
gcode_flavor = reprap
infill_acceleration = 1800
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.68
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 120
interface_shells = 0
layer_gcode = 
layer_height = 0.28
max_fan_speed = 100
min_fan_speed = 45
min_print_speed = 10
min_skirt_length = 0
notes = Used the XY Size compensation in order to make more accurate holes. \nBut this also changes the size of the overall part and in the test with holes and pegs inside, it also changes the outer diameter of the peg.
nozzle_diameter = 0.35
octoprint_apikey = 61BBF4236E784E81AED4115F076DB2D3
octoprint_host = 3ditto1.local
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base]-[timestamp].gcode
overhangs = 1
perimeter_acceleration = 800
perimeter_extruder = 1
perimeter_extrusion_width = 0.38
perimeter_speed = 75
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 2
retract_length_toolchange = 3
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = random
skirt_distance = 3
skirt_height = 1
skirts = 3
slowdown_below_layer_time = 15
small_perimeter_speed = 50
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.5
solid_infill_speed = 75
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = M140 S[first_layer_bed_temperature] ; Set bed temp, No wait\nM104 S[first_layer_temperature] ; Set hot end temp, No wait\nG28 ; Home all axes\nG1 Z50 F9000 ; Z to ready at max\nG1 Y190 F9000 ; Y to ready at max\nM84 ; Motors off\nM104 S130 ; Set hot end holding temp, No wait\nM190 S[first_layer_bed_temperature] ; Set bed temp, Wait...\nM109 S[first_layer_temperature] ; Set hot end temp, Wait...\nG92 E0 ; Reset extrusion distance\nG1 E2.5 F900 ; Start head cleaning move, Extrude filament, Limit up to E max...\nG1 Y95 Z0.35 E12.5 F280 ; Limit to Z max...\nG1 X5 Y45 E17.5 F900 ; Limit up to E max...\nG1 X0 Y5 E21.5\nG1 Y0 F9000 ; Limit to XY max\nG28 ; Home all axes
support_material = 1
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 1
support_material_interface_spacing = 0.1
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 45
temperature = 205
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0.4
top_solid_infill_speed = 60
top_solid_layers = 4
travel_speed = 160
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = -0.15
z_offset = 0

alexrj, 2015-03-23:
Thank you for the detailed report. Also congrats for using the Notes field in Print Settings. :-)

alexrj, 2015-03-28:
Unfortunately it doesn't crash for me. I tested with both 1.2.6 and 1.2.7-dev. Can you reproduce it regularly with the instructions you posted above? 
(Also, which one of the two linked STL files generates the crash?)

tobiasvogel, 2015-04-02:
I'm getting the same error on my Mac OS X 10.9.5 with Slic3r 1.2.6, alltough I am not getting it with the specific model mentioned above, but with several of mine - the critical point seems to be a combination between File-size and Complexity.

I guess the problem is the following (just launched from a Terminal to get the output):

    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    Out of memory!
    slic3r(57183,0xb039f000) malloc: *** mach_vm_map(size=8388608) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    Out of memory!
    slic3r(57183,0xb06ab000) malloc: *** mach_vm_map(size=8388608) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    Out of memory!
--- snip ---

    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    slic3r(57183,0xb0629000) malloc: *** mach_vm_map(size=8388608) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    slic3r(57183,0xb0629000) malloc: *** mach_vm_map(size=8388608) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    slic3r(57183,0xb0629000) malloc: *** mach_vm_map(size=8388608) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    Bus error: 10

To me (not very into Perl though...) this clearly looks like a memory restriction issue, however I cannot spot it on the first sight, as `ulimit -a` returns 'unlimited' where relevant:

    core file size          (blocks, -c) 0
    data seg size           (kbytes, -d) unlimited
    file size               (blocks, -f) unlimited
    max locked memory       (kbytes, -l) unlimited
    max memory size         (kbytes, -m) unlimited
    open files                      (-n) 256
    pipe size            (512 bytes, -p) 1
    stack size              (kbytes, -s) 8192
    cpu time               (seconds, -t) unlimited
    max user processes              (-u) 709
    virtual memory          (kbytes, -v) unlimited

I got 16GB of Ram, so that should be enough I guess...

EricBabin, 2015-04-02:
Hi Alessandro,

Sorry it took me so long to get back to you.
Strange. I just tried it now and it seems to work. I tried with the settings settings I already had in Slic3r and with the settings from this post - just in case they had changed. I'm doing the same thing. I rotated the object and tried exporting the gcode.

The only difference I can think of is that I might have less applications open at the same time as Slic3r and maybe the extra memory is helping? I usually have lots of apps running during my work day.

If I hit this situation again, I will try closing some apps and see if it works.

Thoughts?

Eric

alexrj, 2015-05-04:
I can't really reproduce this. It's interesting because Out of memory is clearly a memory error coming from the system. But your settings and your test case STL are not much complex. It would be nice to know how much memory Slic3r is using when you process that file. You can do that by keeping the Activity Monitor app open and looking at Slic3r's memory usage.

tobiasvogel, 2015-05-04:
Interesting! Watching the Activity Monitor is somewhat informative. The thread continuously keeps blowing up until it reaches ~ 3.26 GB and then crashes with this warning, as before:
[image]

tobiasvogel, 2015-10-26:
In the meantime there seems to be quite a bunch of people having this issue, e.g. issue #3098, #3029, #2992, etc.
As far as I could find out by now, this error is raised as soon as a certain amount of memory for the geometric calculation: this explains why some STLs crash Slic3r and some dont.
The point where the error is thrown is dependant on system memory and vertex complexity of the polygon.
Googling for this error showed that people dealing with large DBs run into this error quite often, and the only proper solution there seems to be handling the data in chunks rather than reading everything into the memory at once..
Maybe this approach would be reasonable for Slic3r, too: get the maximum amount of memory malloc() can allocate and then staying within that limit while processing the geometry in order to be safe from the system killing the application?

adernicolasK8200, 2015-10-27:
The problem I have (issue #3098) might be of the same origin, BUT: 
- Slic3r crashes when loading the object regardless of the complexity. Had the same result with a simple 10mm cube ;
- Slic3r crashes even though it is the only application running - I also tried to run it with all networks OFF (wifi, bluetooth, ...) and tried run after rebooting ;
- I can't technically increase the memory of my Mac, as it's already at its maximum...

alexrj, 2017-03-17:
We now have a brand new rewritten packaging toolchain for OS X.
You can try the automated builds available at http://dl.slic3r.org/dev/
I'll close this issue as it's very old and very likely to be fixed, but feel free to reopen if you need further assistance.

closed 2017-03-17 by alexrj