slic3r/Slic3r #2754: Gcode Export remains disabled after canceling export

OhmEye, 2015-03-24:
Current git master with default settings running on Linux 64bit:

On the Plater tab, add an STL.
Click the Export G-code button.
Click the Cancel button at the bottom of the window.
The Export G-code button remains disabled.
Choosing the Export G-code menu item from the Plater menu give the error: "Another export job is currently running."
Exporting to G-code is not possible without restarting slic3r.

lordofhyphens, 2015-03-24:
It is probably because the threads haven't joined yet.

I can confirm the general behavior though.

OhmEye, 2015-03-24:
It also happens when single threaded. I didn't realize default was 2 threads, my own profiles are all single threaded but I did a check with defaults to make simpler reporting.

alexrj, 2015-03-28:
Fixed, thank you.

closed 2015-03-28 by alexrj