slic3r/Slic3r #2755: Feature Request: Print only specific layer range

larsen0815, 2015-03-24:
I have already proposed this in the RepetierHost forum but it seems this shall be the better place for this request.

Sometimes there are objects with critical sections that are difficult to print and you need to experiment with slicer settings to get it right. You will need some test prints until you find the best slicer settings.

This could be improved, if you could start printing at a specific layer thereby skipping all the underlying layers where you know there will be no problems. At the moment, you would have to manually edit the g-code, delete the first layers and then correct the absolut value for the z-axis.

Let the user specify a layer range and e.g. print only layers 60 to 80.

xoan, 2015-03-24:
http://manual.slic3r.org/expert-mode/variable-layer-height

«An additional feature of the variable layers height option is that by entering a zero for a range that part of the model will not be printed. Fig. shows the G-Code where layers between 0 and 4mm are skipped. This is a useful way of dividing a tall model into multiple, shorter sections which can be printed individually and assembled afterwards.»

larsen0815, 2015-03-24:
Thx, this will be useful for now. Though, this is only possible with standalone slic3r and not when called by RepetierHost, isn´t it?

lordofhyphens, 2015-03-24:
You could also just cut the model twice with 1.2.x (experimental). Even
easier if you are trying to get something right. Then export the plate and
you can do whateveryou want with rephost (which I don't bother with, but
that is a different story).

Might be straightforward to adapt the current cut ui to let you do both
cuts at once.

lordofhyphens, 2015-03-24:
Also layer isn't something you know aprori without also knowing the layer
heights. I am a fan of making the variable layer height stuff easier to use
(more visual and less math).

larsen0815, 2015-03-24:
Currently I would slice once to check the visual presentation and get the layer number as shown in RH. With a second run I could then define at which layer to start.

lordofhyphens, 2015-03-24:
If you know the absolute z value, you can cut the model.

This has some side effects though.  Your current solution is probably the
best. Then again it would be easy to write a small shell script  (in
windows land vbscript ,jscript or a batch file) that trims the file for you
from/to a given z height.

lordofhyphens, 2017-03-20:
@larsen0815 Use Slic3r's plater, it's much better at this point than rephost :)

Also, there is host functionality built into Slic3r now for sending gcode.

luisrosety, 2018-01-11:
I am looking for a way to generate gcode for printing first layer(s) for test purposes.
As first layer is very important, being able to generate only first layer test gcode would help a lot for testing different layer height, width, speed, flow etc.
I know that cutting above a given Z is possible, but it would be better for test printing just a few layers.
Any ways to do it?
Thanks

lordofhyphens, 2018-01-11:
Just turn on verbose gcode and put together a small script to drop layers.

lordofhyphens, 2018-01-11:
Put something distinctive in your end gcode to keep it around, like `; begin end gcode`  and turn on verbose gcode (see Print Settings -> Output Options). 

Save this script somewhere and add it to the post-process gcode list (it *should* work , but I haven't tested it at all; users are responsible for debugging). 

````
#!/usr/bin/perl -i                                                     
use strict;
use warnings;

$skip = 0;
while (<>) {
     if ($skip == 1) {
         if ($_ =~ /\Qbegin end gcode\E/) { 
         $skip = 0;
    } else { next; }
     }
     $skip = ($_ =~ /\Qmove to next layer (1)\E/);
     print;
}
````

luisrosety, 2018-01-12:
Thanks, it works!