slic3r/Slic3r #2756: Feature request: Seperate speed for first layer extruded onto support

kyrreaa, 2015-03-25:
I have discovered a very cool feature of my new strong hotend.  I can extrude plastic that is in the phase transition temp and that will not bond well to layer below, but will adhere to it nicely for print.
It requires a higher extrude rate than normal for a given temperature.

If Slic3r could specify a different printspeed (which could be exempt from auto cooling speed adjustments) we could put down first layer so it will not bond to even solid printed support surface.
This will make large flats printable with full support without them adhering too much.
The next layer would of course be printed normally to make it bond to the previous layer.

The setting must of course not apply to all the printing in the layer, only the parts printed ontop of support material.

lonelymyp, 2015-04-21:
The latest version of the slicer has a super option: "Z distance between object and support material"
Using air cooling, it comes off very easily.

kyrreaa, 2015-04-21:
The Z distance option is competely the opposite of what I am trying to accomplish here.
My manual hacks allow me to print perfectly flat bridges with <b>smooth</b> bottom and still have them come off directly. This works by controlling the bond between the extruded plastic and previous layer without reducing contact surface like Z distance does.
My method is a replacement/alternative to using a soluble support material but works best on large areas.

lordofhyphens, 2017-03-20:
I'm not sure this is actually what you want, as you'll probably have curling if you managed to implement it.

Martronic-SA, 2017-06-05:
The idea is good especially for support materials like scaffold where the adhesion of supported material can be a bit poor. A lower speed for supported part might help to maintaint the part in place (like lower speed for the first layer)