slic3r/Slic3r #2757: Feature request: optional printing of the "outside" support material only

Zuborg, 2015-03-25:
When I print complex mechanical parts often need a support material under them. It prevent a bend and vibration of tall and inclined models. (I prefer a tilt angle near 45 degrees) But when I adjust appropriate "overhang threshold" for generation the support from printbed up to the model it also generated unnecessary support in internal hollow volumes of the model. It will be a good selectable option to generate support when it can reach printbed directly and disable support for other cases (internal and shielded by the model body).

lonelymyp, 2015-04-21:
It would be just super function.

I recently printed a salt shaker in the form of a pig, and got the problem of removing the support from the inside. 
Pillar of support was larger than the diameter of the hole.

Support within the model is not needed, no one will see if several perimeters fail inside. 
But without the support, the snout is not printed

lordofhyphens, 2017-03-20:
#3310

PR #3525 (2017-03-26): A new feature "support_material_buildplate_only" implemented.
Also a bug has been fixed for zero interface layers. Before
slic3r would put infinite number of interface layers over top surfaces,
if the number of interface layers was set to zero.

Fixes #3310 
Fixes #2757
Thanks to @bubnikv for this work.

Win32 automatic development build: https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-infill-over-plate-192#files
closed 2017-03-20 by lordofhyphens