slic3r/Slic3r #2758: Slic3r / Perl consumes all memory

c-born, 2015-03-25:
I've just updated to the current head 2d7bfbb805b9e93a3d384b2b4e10f56bf2f33966  to test the fix for Issue #2691, and I'm happy to report that it seems to be fixed. I just did the two thin-wall test objects, as per my last image in that issue, and checked the G-Code was ok. It was, although those strange diagonals are still there.

However I left Slic3r running overnight, and next morning my system was acting very bogged down and constantly paging. I found that Perl/Slic3r was using over 6.5GB of RAM, more than half my total. It had not been touched after generating the G-Code for the test.

Win7/64, conditions, settings etc are all still as in Issue #2691

eras, 2015-03-26:
I just yesterday updated Git from a very old version, left it idling overnight and noticed that at night the Linux kernel had OOM-killed it. Version 2d7bfbb.

Curiously my process too died at around 6.3 gigabytes, though I have 32 gigabytes of it and at the moment 10 gigabytes in disk cache, quite a coincidence.

eras, 2015-03-26:
Actually it may very well be possible that Slic3r had increased to that size when I had been using it and the OOM chose it to be killed due to its size. But still, quite interesting.

I noticed that Slic3r virtual memory usage increased from 500M to 1.5G just loading a 400k stl (and then it probably slices it automatically). Deleting the model and reloading it didn't increase memory usage.

alexrj, 2015-03-28:
I left Slic3r open on OS X overnight and it didn't grow. It's still at 50Mb. I'll test it on Windows...

c-born, 2015-03-28:
Testing again myself now with the current build. Same input, I just can't recall which screen I had left it on last time, if that could make a difference?

klpeders, 2015-04-07:
The other day I had to kill Slic3r because the system had become very unresponsive:
>    PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
>    19004 kp        20   0 13.033g 0.012t  27584 S   0.0 79.1  11:48.81 slic3r  
(that is 13GB / 80% of the RAM accumulated during a period of 12hours).

This is on 1.2.6 prebuild on linux 64bit.
I have been unable to reproduce the automatic leak problem, but it seems quite easy to provoke few other memory leaks.
1. I loaded a big project: [BeQui](http://www.thingiverse.com/thing:701051/#files)
2. Drag all the parts to the Plate. Memory consumption to go from 800MB (fresh start) to 2GB.
3. Press Arrange button a few times. An additional 200MB is consumed at every click
4. Now switch to 2D mode. Click on a part to select it and notice that memory increases ~250MB on each select.
5. Now press Delete All to get a fresh plate (notice that no significant amount of memory was released - here, the memory is down to ~4GB).
6. Drag the files to the plate again - the memory consumption end at 6GB.

Lenbok, 2015-05-10:
I have had this several times on Ubuntu x64 14.04, 901716a. I had sliced a print, uploaded to my octoprint, and leaving slic3r alone the memory use grew over time until all OS memory was gone.

Lenbok, 2015-05-10:
I just did a little experimentation to try and narrow this down and for me it is 100% reproducible.

Moved my ~/.Slic3r out of the way.
Start slic3r
Click Next Next Finished to get through the config wizard with all defaults.
Drag an stl (anything works for me) onto the default view (3D)
Wait for the auto slicing to finish. 
At this point everything is stable.
Now click on the Layers tab (nothing else is required).
Watch as one CPU becomes permanently active and memory use grows unbounded.

Lenbok, 2015-05-13:
$ git bisect bad
25cddfe446c329a4808fe6dcbf87722a5708bb8c is the first bad commit
commit 25cddfe446c329a4808fe6dcbf87722a5708bb8c
Author: Alessandro Ranellucci <aar@cpan.org>
Date:   Mon Mar 9 15:30:19 2015 +0100

    Bugfix: layers view was not resized when inactive. #2608

gege2b, 2015-05-13:
Just tried this too
and in fact, it eats approx. 64 KB/s while using 4% of CPU time
Right now, he uses 160 MB, I'll see how many after lunch

Win7 x64 
current dev compiled with citrusperl x64

gege2b, 2015-05-13:
Hop !
Just came back from lunch
Slic3r (well, "perl.exe"), is now using almost 338 MB

Lenbok, 2015-07-15:
FWIW, the released 1.2.9 does not exhibit this problem for me, so either it is fixed now, or related to running a build from git rather than a release version.

closed 2017-07-10 by c-born