slic3r/Slic3r #2761: gcode generation: corrupt walls with 'Spiral vase' enabled

klpeders, 2015-03-28:
The generated gcode showing strange cross instead of a thin wall.

## Slic3r version 
Slic3r 1.2.6 Prebuild binary for Linux 64bit

## Operating system type + version
Fedora 21, Intel 64 bit

## Steps to reproduce the issue, including:
* Generate STL in OpenSCAD:
    $fs = 0.1;
    $fa = 5;
    union() {
        linear_extrude(height=8) scale(7)
            text("6", font="serif:style=Heavy");
        linear_extrude(height=5) scale(2.8) translate([17.5,2.2,0])
            text("号", font="Source Han Sans CN:style=Heavy");
    }

* Expected result
[image]

* Actual result 
[image]

## Slic3r configuration file 

I had to add .gif to be able to upload the config and stl files:
[image]

[image]

alexrj, 2015-03-28:
I was able to download your config, but the STL file is corrupt. I guess GitHub processes it as an image file. Can you host it somewhere and put the link here?

klpeders, 2015-03-29:
It works for me:

    $ wget https://cloud.githubusercontent.com/assets/11691683/6880344/dbd738f6-d566-11e4-87a4-f058eaa721e3.gif -O no6.stl.xz
    $ unxz no6.stl.xz
    $ ./bin/slic3r no6.stl.stl 
    => Processing triangulated mesh
    [...]
    => Exporting G-code to ./no6.gcode
    Done. Process took 0 minutes and 1.245 seconds

alexrj, 2015-03-29:
You didn't tell me it was XZ compressed...

alexrj, 2015-03-29:
Okay, thank you, I can reproduce this.

alexrj, 2015-04-12:
Fixed. This model is not suitable for spiral vase, and Slic3r wasn't detecting this correctly (anymore).

closed 2015-04-12 by alexrj