slic3r/Slic3r #2762: [1.2.7-dev] Part/Object settings window incorrect initialization

Lenbok, 2015-03-29:
When I open the part settings it initially looks like this:
[image]
I would expect that straight after opening this window I should be able to start adding settings. After all, the '+' icon is present. However, this does not work.

In the tree view it looks like it should currently be in "Object Settings" mode, but the settings area below is labelled "Part Settings". Note that this settings area no section for the Extruder. Clicking on the '+' icon does not bring up a list of settings that can be applied.

If you click on the part name (in this case vacuum-pipe-lock), the "Extruders" section appears and per-part settings can now be chosen via '+'. 
[image]

Clicking on "Object" to switch to "Object Settings" now contains the missing Extruders section and object settings can now be chosen via '+'.
[image]

It seems like this window is being initialized incorrectly so it isn't completely in either Object Settings or Part Settings.

alexrj, 2015-03-30:
Good catch...

alexrj, 2015-03-30:
Fixed. Thank you!

closed 2015-03-30 by alexrj