slic3r/Slic3r #2763: feature request: modifier meshes when running headless

mattratto, 2015-03-30:
I'd like to be able to call the modifier meshes functionality when running slic3r headless. In the design process I will predetermine areas requiring different settings, generate meshes encompassing those areas using a solid modeler and using generic names, and then call them out from ini file.

alexrj, 2015-05-03:
Slic3r supports that; you need to use the AMF file format.
In order to see how it works, try to define a multi-part print with the GUI and then export the AMF file. Open it and see how it's made.

mattratto, 2015-05-06:
great! Thanks so much Alessandro. Slic3r is an essential tool!

lordofhyphens, 2017-03-19:
As the man said, it's already implemented, just have to do it with AMF.

closed 2017-03-19 by lordofhyphens