slic3r/Slic3r #2764: Feature Request: Adjustable minimum segment length

NitroXpress, 2015-04-01:
Is it possible to make the minimum segment length for arcs respectively circles adjustable ?
Shorter segments contribute to better trajectory generation and smoother surfaces.

Greetings

Frank

lordofhyphens, 2015-04-01:
So polyholes? Can't you fix this in the model?

alexrj, 2015-04-01:
Slic3r does not generate segments from arcs, since the input models (STL files) don't contain arcs. Slic3r generates G-code having exactly the same number of vertices as the original STL file contains - no loss of detail. If you want smoother curves, you need to export your STL files with more facets from the CAD itself.

NitroXpress, 2015-04-01:
Hi,

I know, in .stl models are no arcs.
But don't know how to explain it exactly...

Slic3r doesn't follow the original number of vertices.

I've created a 50mm diameter cylinder with a angle resolution of 1° (360 segments).
But Slic3r only outputs 128 segments.

Did I something wrong ?

alexrj, 2015-04-01:
Have you read the guidelines before posting this issue? Please do.

NitroXpress, 2015-04-01:
Yes, but don't know how to attach a .stl file...

alexrj, 2015-04-01:
Host it somewhere and put the link here.

NitroXpress, 2015-04-01:
Much effort...
Can I email it direct to you ?

alexrj, 2015-04-01:
No, please.

NitroXpress, 2015-04-01:
OK.

Too much bureaucracy for me here (like in Germany)

Bye...

alexrj, 2015-04-01:
I do this on voluntary basis. This is not a company help desk. Your help is appreciated. Otherwise, goodbye.

alexrj, 2015-04-01:
(And writing Slic3r for you was "much effort" as well, by the way. lol)

closed 2015-04-01 by alexrj

NitroXpress, 2015-04-01:
Yes, i know that all...

And i have great respect about your great work !
No  reason for gallows humor...

I want to help making Slic3r more perfect.
But how to, if not so ?

a4jp-com, 2015-04-01:
You can upload your files onto any upload site. I did a quick search and found this: http://www.filedropper.com/

lordofhyphens, 2015-04-01:
NitroXpress,

Uploading the test case allows people other than Alessandro to test as
well. It's just better practice overall. You can help by reporting issues
in the proper way, or modifying Slic3r so that it does what you want and
submitting a pull request that explains the reason for the change and why
it's important.

Adding your own changes should not break any existing tests either and
should be readable by other people.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

NitroXpress, 2015-04-01:
Damned...
I am forgetful :-(
You can download the file in the RepRap forum.
The file is attached there.
Scroll a little bit down...
http://forums.reprap.org/read.php?263,490722

NitroXpress, 2015-04-01:
Ok.
Shame on me !
I am very sorry.
That was only a picture.

I will find a nother way...

lordofhyphens, 2015-04-01:
2shared, filedropper, etc.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2015-04-01:
But seriously, read Contributing.md before you post any more.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

a4jp-com, 2015-04-01:
NitroXpress, thank you for reporting the bug ^^

NitroXpress, 2015-04-01:
...
I don't know if this a bug...

@lordofhyphens 
But thanks for the hint... 
Contributing should not be the problem.

NitroXpress, 2015-04-01:
Sorry again...
It was the wrong thread in the slic3r forum.
Now the right link:
http://forums.reprap.org/read.php?263,490228

The file is attached at the third post.

a4jp-com, 2015-04-02:
All of our config files are different, so you need to upload your individual config file to get help. Your settings just might be wrong.

NitroXpress, 2015-04-02:
I have the highest resolution setting in slic3r with a extrusion wide of 0,25mm.
What can i do more ?
Have you downloaded the files from RepRap forum ?

I found out, that decreasing the resolution of my CAD output to 2° (180 segments) slic3r forces to create 188 segments.
Don't know why...

Both files can be downloaded in the thread here:
http://forums.reprap.org/read.php?263,490228

a4jp-com, 2015-04-03:
I only see an STL file...

a4jp-com, 2015-04-03:
http://www.filedropper.com/

NitroXpress, 2015-04-03:
What file format would you have ?
.stl is standard for 3D printing, or not ?
I tried with many CAD programs, same problem every time.

Sorry, I don't want to share my settings with the whole world.

I asked Alessandro to direct email the files.
I don't understand what the problem is thereby.

Nevertheless, thanks for your effort.

Now back to CURA...

xoan, 2015-04-03:
If you want to other people help you, other people need to be able to
reproduce your exact environment, so you need to share all the information
you have.

If you don't want to cooperate, you can't demand anything.

And if you are comfortable using Cura, please, keep using it.

Thanks.
El 03/04/2015 18:04, "NitroXpress" <notifications@github.com> escribió:

NitroXpress, 2015-04-05:
Hi,
i am not married with Cura...
Currently i am using 3 Slicers. (Skeinforge, Slic3r, Cura)
Each has advantages and disadvantages.
My goal is, to have 1 slicer for all cases.
I think slic3r is the most promising...
There is only 1 problem: resolution.

My hope was, to modify the source code for higher segment resolution.
But no success, although i dug in the source code for many hours.

Sorry for the trouble !

Frank

lordofhyphens, 2015-04-05:
Well yeah, segment resolution is a *model* problem. If you want more
segments in an approximation of an arc,  then fix your model. Slic3r would
have to able to read your mind otherwise.

NitroXpress, 2015-04-05:
No, it is not a model problem !

I have made the cylinder (50x10) with up to 300000 facetes.
But slic3r output is only 128 segments on circumference.

lordofhyphens, 2015-04-05:
Please post a link to your test cylinder  and your settings then.

NitroXpress, 2015-04-06:
config

http://www.filedropper.com/config_10

NitroXpress, 2015-04-06:
Thanks for you considerations.
But no difference...