slic3r/Slic3r #2765: Strange lines.

a4jp-com, 2015-04-02:
There are weird lines in the windows of a small house that I made. Is this a bug?

[image]

http://a4jp.com/3d-files/bugs/2015-04-02nd-01-42-pm-for-github-config.ini

http://a4jp.com/3d-files/bugs/2015-04-02nd-github-1-apartment.stl

lordofhyphens, 2017-03-19:
I think it's just a draw artifact.
[image]

a4jp-com, 2017-03-20:
The windows kind of get filled in when I print so Slic3r might be calculating something. Any way to be sure?