slic3r/Slic3r #2766: Slider is not updated or reset to original full position.

a4jp-com, 2015-04-02:
Hi @alexrj 

If I load a model then move the slider everything is okay. But when I press delete all and load another model the slider is in the same position showing everything. When I click on the slider's handle it redraws the model at the correct layer. 

[image]

http://a4jp.com/3d-files/bugs/2015-04-02nd-01-42-pm-for-github-config.ini

http://a4jp.com/3d-files/bugs/2015-04-02nd-github-1-apartment.stl

lordofhyphens, 2017-03-19:
This looks fixed as of 90395f93940410093244e9cfcaf3dce407ddc811

closed 2017-03-19 by lordofhyphens

a4jp-com, 2017-03-20:
Just checked. It works properly now ^^