slic3r/Slic3r #2767: Feature Request: Reduce Warp Feature/Setting.

corbycar, 2015-04-02:
I have found that putting holes near warp prone areas of a print breaks up the pull of the shrinking layer, allowing prints that would normally warp to be printed successfully. If we could just omit a few random g code lines in these areas, or have an option to create random "gaps" in each layer near corners and ends of long/ wide prints (I believe) it would function as breakers that would allow shrinking without (or at least with less) warping. this would solve many problems for many people and make 3d printing more user friendly. I do not know  if this is a reasonable request, or how difficult this would be to program, but you guys are geniuses so I am requesting with high hopes. I envision a variable setting that would allow from no change, to making Swiss Cheese out of a prints corners. can it be done?

lonelymyp, 2015-04-21:
It would be just super function.

I recently printed a salt shaker in the form of a pig, and got the problem of removing the support from the inside. 
Pillar of support was larger than the diameter of the hole. 

Support within the model is not needed, no one will see if several perimeters fail inside. 
But without the support, the snout is not printed

lordofhyphens, 2016-05-29:
Post-process script would probably be able to sort it out.