slic3r/Slic3r #2768: [1.2.6 experimental] modifier mesh seam

curiouspl2, 2015-04-03:
see the picture - in the very center, this prints exactly as on preview (which is very cool actually as it shows how accurate new preview is - big thanks to Alex for good work :) :

[image]
 
no idea why those two perimeters create such gap, but it looks quite preventible (they could overlap a bit more perhaps?). at first i've thought it's printer calibration issue, but then i've seen it prints exactly as on preview , and increasing flow rate just makes it worse (blobby) :)

to reproduce it's enough to snap any two overlapping objects together choosing other as modifier mesh, then modify any perimeter setting . 
on the pic perimeter extrusion widths are 0.65 and 0.7 , but seam is created on all settings. 

if you want to use files i've used, you may try those : 
http://www.thingiverse.com/download:1209532
http://www.thingiverse.com/download:1209533

or this one : 
https://github.com/alexrj/Slic3r/issues/2499

lordofhyphens, 2015-04-12:
I'm wondering if the mechanics of infill/perimeter overlap changed at some point so the % listed isn't what we think it is.

curiouspl2, 2015-04-19:
this is not infill/perimeter issue , it's perimeter/perimeter issue - collision avoider perhaps.

i've found temporary workaround for that, 
check out screenshot : 
[image]

http://www.thingiverse.com/download:1248879
http://www.thingiverse.com/download:1248877

to put it brief : first you make your reinforcements, then you boolean subtract them from original file (i.e. in blender or meshmixer), export the files with holes, and then export reinforcements alone. 

then open all files up as on screenshot - reinforcements as object instead of modifier. 
this way collision checking is gone, and you can even prepare your reinforcements to overlap a bit (like i did) with model to make them bonded with model by i.e. scaling, moving , inflating etc.