slic3r/Slic3r #2770: [1.2.7-dev] Combination of volumetric E and pressure advance causing huge extrusions

lordofhyphens, 2015-04-04:
For the model at https://www.youmagine.com/designs/5mm-calibration-steps-a5000a11-90b8-4664-aa28-add949efbd85, I turned on volumetric E and pressure advance and immediately saw an E move of ~45 in the gcode, and some retraction moves as well (which were enough to back the filament all the way out of the extruder!) This didn't happen (at least in not such a way that I could notice) without volumetric. The extrusions are causing a huge blob on the print, and the retractions are enough to kick it out of the extruder after the second one (on this print, it doesn't print enough to re-prime the head).

Could be a use-case related to #2501  or #2470, specific to volumetric E. 

My filament diameter is set to ~1.1 as per the instructions on the Smoothieboard wiki (which I am using), http://smoothieware.org/extruder.

Here's a copy of the config, increasing pressure advance made things worse.:

# generated by Slic3r 1.2.7-dev on Sat Apr  4 02:28:49 2015
avoid_crossing_perimeters = 1
bed_shape = 0x0,225x0,225x225,0x225
bed_temperature = 110
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 10
complete_objects = 0
cooling = 1
default_acceleration = 2400
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = G10\nM104 S0\nG91\nG1 Z1\nG90\nG1 X0 Y0\nM84     ; disable mots\nM140 S0
external_fill_pattern = concentric
external_perimeter_extrusion_width = 0
external_perimeter_speed = 80%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.12838
fill_angle = 45
fill_density = 25%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 110
first_layer_extrusion_width = 1
first_layer_height = 0.26
first_layer_speed = 20
first_layer_temperature = 235
gap_fill_speed = 30
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 2400
infill_every_layers = 3
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 1
infill_overlap = 115%
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 100
min_fan_speed = 100
min_print_speed = 10
min_skirt_length = 5
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 543BC61C6D3240F78130EA181149B6F6
octoprint_host = 192.168.1.104:5000
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 2400
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 3
post_process = 
pressure_advance = 1
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 0
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 11
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 5
small_perimeter_speed = 30
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 80%
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = M106 S0\nG28 Z0 ; Home Z\nG90; set to absolute position\nG1 Z0.1; leave some room for oozing\n\nM140 S[first_layer_bed_temperature]; Wait until heatbed hits [first_layer_bed_temperature] \nM109 S[first_layer_temperature]; heat nozzle to [first_layer_temperature]C;\n; wait 2m?\nG1 Z5 F100 ; lift nozzle\nG11
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 1
support_material_interface_spacing = 0.05
support_material_interface_speed = 30
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 30
support_material_threshold = 0
temperature = 235
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 80%
top_solid_layers = 3
travel_speed = 80
use_firmware_retraction = 1
use_relative_e_distances = 0
use_volumetric_e = 1
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0

alexrj, 2015-05-03:
I think you just need to reduce your Pressure Advance factor...

hakimio, 2015-06-22:
@alexrj
No, it also happens with small pressure advance K factor (0.01) and without volumetric E (I was able to reproduce this with 1.2.7 release version). There is some serious bug in pressure advance logic which needs to be fixed.

alexrj, 2015-06-22:
@lluis, can you help here?

alexrj, 2015-06-22:
By the way, @lordofhyphens if you use volumetric extrusion it's useless to set Filament Diameter to 1.1. You should still set it to the actual filament diameter, since the K constant depends on filament diameter. Thin filament requires smaller K constants. So if you get too high E values in G-code, reduce the K constant.

hakimio, 2015-06-22:
@alexrj 
The problem is that those huge extrusions happen randomly in the middle of the print. At first it prints normally and then out of a sudden print head movement stops and it starts extruding huge amounts of plastic in a single spot.
If it was an issue with K constant set too high, those huge extrusions would just appear while printing the very first layer.

lluis, 2015-06-22:
s/ @lluis / @jluis /

hakimio, 2015-06-22:
It should actually be @llluis (3 * "l").

alexrj, 2015-06-22:
Whooops, sorry @llluis and @lluis!

llluis, 2015-06-22:
Here I am! Sorry for being away, but a child can be very time demanding sometimes.  :)

Last time I checked, your implementation of Pressure Advance was broken, as I mentioned here https://github.com/alexrj/Slic3r/commit/ff9b53260d6084fa699d1c79c0f73c5c292a488a, but I didn't have time to look at.

Will try to do this this weekend.
:)

llluis, 2015-06-22:
Saw a few bugfixes in the feature after my last check, so indeed I need to look at it more carefully.

hakimio, 2015-07-04:
@llluis @alexrj 
I think I have found out what's causing huge extrusions in a single spot when pressure advance is enabled. It seems to be caused by high acceleration (or big difference in acceleration/speed).
Here is what I have tested:
1. At first I've printed knight Andreas (the STL file included at the bottom of the post) reduced to 50% of the size and bridge acceleration was set to 1000mm/s^2. There were no issues.
2. Then I enlarged the same model to 75% and increased bridge acceleration to 3000mm/s^2. This time in the middle of the print it decided to stop moving and extrude a large amount of plastic in a single spot.

I've used "Repetier Host" to change model size and then slic3r 1.2.9 cli to slice it. I have uploaded slic3r config, STL model, gcode and photos of the failed print [here](https://www.dropbox.com/sh/xovsio63z0x6sgo/AADcNzXQcXBJfH2Qvo3Fc2VEa?dl=0). You can see an ugly stripe in the printed model where huge extrusion occurred.
Tell me if you need any additional information.

hakimio, 2015-08-19:
@llluis @alexrj 
CapnBry found what's causing those huge extrusions (see issue #2960):
> The move from Y103.513 to Y97.75 = 5.763mm and extrudes 1.0277mm of filament. This is 1.75mm > filament so at 0.24mm layer height that's a 1.6mm wide bead on both sides or 4.1x the width of the > > gap. This might go unnoticed but I have pressure advance on, so to prepare for this giant extrusion > > move, it pressure advances over 18mm of filament at 30mm/s which doesn't really work all that well, > then it tries to pressure release the same amount which immediately jams the E3DV6 extruder.

a4jp-com, 2015-08-20:
Nice work @hakimio and @CapnBry

lordofhyphens, 2016-03-27:
Okay, at least some of my issues is related to my use of Smoothieware, which treats all of E moves as volumetric if that's enabled (as opposed to Marlin). This would apply to retraction moves as well.

I will look into adding yet another fw variant (sigh).

lordofhyphens, 2017-03-25:
It looks resolved as of current master (I'm not seeing overly large values for advance), and I don't actually use advance or really anything in this config anymore.

closed 2017-03-25 by lordofhyphens